Openvario/meta-openvario #176: Use XCSoar's audio vario instead of variod

MaxKellermann, 2022-01-13:
I saw there's variod, but I don't understand why it exists. XCSoar has an audio vario, and variod is a duplicate effort.
If variod has features that are missing in XCSoar, it should instead have been added to XCSoar; now it's too late, but we can still copy those missing features. Are there any?
If a standalone vario daemon is needed (for whatever reason I cannot imagine right now), we could add such a daemon to the XCSoar repository, based on the existing audio vario code.

linuxianer99, 2022-01-13:
That variod was invented by Andreas Wasserbauer as we had some problems with XCSoar vario function at the start of the project.

I forwarded him the issue via Email, so maybe he can give a bit more details...

In general, i agree with you, that all the function should be integrated into XCSoar as all platforms can benefit ...

herr-ypsilon, 2022-01-13:
Hi Max, 
Variod was written almost  ten years ago. We wanted to have basic vario functionality to work independently of the front-end. In addition XCSoar's Audio Vario wasn't too useful at the time, as the update rate of the whole Vario mechanism seemed to be synchronised with map update (correct me if I'm wrong on this one) which was operating at some 2 Hz refresh rate. Not enough for  convincing vario output. We also felt, that the audio vario's latency is more critical than the front end's.

So you see there's a bunch of thoughts that led to the design. If you think the aforementioned problems are easily (or already) addressed with XCSoar now, then I personally don't mind if variod is replaced by XCSoar's audio vario. Variod is a quick hack, and there's a lot of room for improvement, however whatever that improvement is: Stability of the audio vario is of paramount importance, and should not be compromised.

mihu-ov, 2022-01-13:
Yes, back then during the first flight tests with XCSoar audio vario we had pure sine waves with 2Hz update rate. Not many could cope with that and variod was a huge - and quick - improvement with minimum latency.
variod is also a very modular approach, working with standard XCSoar, LK8000 / Tophat or something else as well. We probably also considered it as a kind of intermediary step to a later hardware solution (like xcvario, Freevario today).
I still like the modularity of variod. XCSoar audio vario would remove the *.xci interface kludge between stick remote -> XCSoar -> variod which would be a big benefit. But I would not sacrifice minimum latency for that (don´t know if there is an effect).

MaxKellermann, 2022-01-13:
No, XCSoar's audio vario is *not* synchronized with map updates at all. It is updated here:
https://github.com/XCSoar/XCSoar/blob/65aacd60dbe019343cb5db4454dc3eae83e861fb/src/MergeThread.cpp#L115-L118
The `MergeThread` is woken up whenever new data is received from any device, but it is throttled: it is invoked at most 20 times per second (`period_min=20ms`), but each invocation is delayed by 10ms (`delay=10ms`). These `WorkerThread` parameters should reduce the CPU load, because more than that is rarely ever useful; and the delay aims to merge adjacent sensor updates into one run.
After each wakeup, it collects all sensors from all devices, merges them into one large `struct NMEAInfo`, calculates missing values via `class BasicComputer` (e.g. vario from baro altitude), and in the end submits the vario value to the audio vario, and the whole `NMEAInfo` object to the `CalculationThread`, where more expensive calculations will be done (task progress, scored distance etc.), and only *after that*, the map will be redrawn.

So the latency you observed may exist (for other reasons, which nobody knows as of now), but it is fixable.

As I said, it has always been possible to rip out the audio vario code into a separate program; I think my largest achievement in the XCSoar code base is to make it modular, by eliminating global variables. Almost any library can be separated. For the audio vario, that has already happened almost 10 years ago, 3 years before the variod project was created. Check:
https://github.com/XCSoar/XCSoar/commit/b5848b648057f2939ddb8b60780f784de4c9a449 (2012-05-29) - note that this was the commit which created the audio vario, and the same commit already contains the audio vario as a separate program (https://github.com/XCSoar/XCSoar/blob/master/test/src/PlayVario.cpp). This was how I developed the audio vario! Until the feature was complete, I never actually ran it inside XCSoar.

This is the very first variod commit, 3 years later: https://github.com/Openvario/variod/commit/4dcbea78296711b8a45c61570c7decf802ef91bf (2015-07-26)

Which is really unfortunate, because all what was needed to make a standalone vario daemon had already been there for 3 years. I wish that developer time had instead been spent on building something new.

hpmax, 2022-01-14:
FWIW, I don't know the history of it, but I do

morihab, 2022-08-30:
I think the reason why people don't use the xcsoar vario for their OV even though it is available is, that to my knowledge, it is not possible to define a hardware mute button vor xcsoar. Or is this even possible?