Openvario/meta-openvario #179: Put firmware in initramfs

MaxKellermann, 2022-01-13:
Rough idea: instead of having a boot and a root partition, have only a boot and a data partition.
The boot partition contains the usual stuff (kernel, device tree), but also an initramfs image. On boot, the kernel loads the initramfs and decompresses it, and then continues booting with this initramfs mounted on root (and can then mount the data partition).

Disadvantages:
- the initramfs is compressed and thus small (less than 50 MB), but may still be too slow
- modifications are not persistent; though for power users, there could be an option to use [overlayfs](https://www.kernel.org/doc/html/latest/filesystems/overlayfs.html)

Advantages:
- dead simple updates (just copy new files to the boot partition)
- data partition is persistent, no data ever lost; clean separation between data and software
- modifications are not persistent (no breakages)

kedder, 2022-01-13:
Something we should not lose in the process is the ability to install / upgrade software over internet. `opkg upgrade` is a very simple way to get updates and it is much more convenient than carry usb stick with the new image all the time (if that's the only way to upgrade). 

In practice kernel updates are very rare, but applications (xcsoar, ov-shell, compman, syncthing) release more often. Rebuilding images every time is not practical. OTOH package-based upgrades are already well supported by the OS.

kedder, 2022-01-13:
I disagree with that. All desktop linux OSes are upgraded via packages. With proper tools downloading and installing packages are simple for "normal" users. ov-shell provides such a tool, it works. Furthermore, it allows me to host a PPA with extra packages for openvario (and make ov-shell actually installable without incorporating it into the base image). Let's not lose this use case.

MaxKellermann, 2022-01-13:
I think your idea of a "normal user" is very different from mine. My "normal user" does not want to host a PPA with extra packages, will not want to download and install packages anywhere, doesn't want to configure Wifi for his glider, has no desktop linux, does not want to answer debconf questions about which daemons shall be restarted after a glibc or OpenSSL update, and he doesn't even want to use ov-shell for all of this.

kedder, 2022-01-13:
Well, my "normal" users are as important as your "normal" users though. My "normal" users want to receive new xcsoar or ov-shell on their OV devices when new release is out, the same way it happens on their phones. Without downloading files to usb sticks, carrying them to the airfield and copying files to the correct locations.

DanD222, 2022-01-13:
I would have to agree with @kedder here - I actually find the updates afforded by ov-shell quite comfortable (connecting a USB-tethered phone to OV and navigating to a menu item in ov-shell is preferable to me over flashing a USB stick).

kedder, 2022-01-13:
Well, all I'm asking is not to lose the ability to install packages. If the partition with the system software is readonly or ram-disk based ("modifications are not persistent") and installing/upgrading packages is not working, that is a deal-breaker.

MaxKellermann, 2022-01-13:
That could be the difference between "user" and "developer" image (https://github.com/Openvario/meta-openvario/discussions/181): users have initramfs and simple&robust updates, and only a persistent data partition; developers have a full root partition they can tinker with, but updates are more complicated (package based; a new image would destroy all existing data, because it's on the root partition).

MaxKellermann, 2022-01-13:
That, again, is orthogonal to the issue here, and misses the point. Using ov-shell would be possible with a user image.

MaxKellermann, 2022-01-13:
An initramfs is writable by default (though I'd prefer to remount it read-only), so you can install all the packages you want, but they're lost after a reboot.
I doubt that normal users will ever install new packages. All they want is get the latest updates, and that's where this idea shines.

linuxianer99, 2022-01-13:
Yes, agree with your point.
So my suggestion would be, to have both variants:

- Image for 90% of the users, who just wants to have the latest tested software. => Updating the initramfs as blob
- Image for 10% of the users (power users and delevopers) => having a writeable rootfs as it is now, so package update of individual software components is possible => easier test of beta software in flight, etc ...

I am not expert of initramfs, but
@MaxKellermann : you mentioned the possibility of overlays ?
Or is there a way, that maybe XCSoar binary can be stored on the data partition as well and be mounted if newer than the version included in the initramfs ?

at all: Should we move these discussion also to the discussion tab ??

MaxKellermann, 2022-01-13:
Yes, overlayfs could be used - though overlayfs is documented to be fragile if a lower layer gets modified (e.g. by updating the initramfs). I don't know what that means, I've never used overlayfs.
The data partition should be mounted `noexec`, and I don't like the idea of putting executables on the data partition; this doesn't solve the problem (still can't install packages), but creates new ones.

kedder, 2022-01-13:
> I doubt that normal users will ever install new packages

Not only they _will_ install new packages, they are already doing it. The only problem is that right now the procedure is very clumsy and not friendly (involving command line). What we _can_ do is to make it more accessible to users.

MaxKellermann, 2022-01-13:
Andrey, you are just repeating yourself, and I could go on all day repeating myself that the users you are talking about are not "normal" in my definition, but ... let's both save the time and do something productive instead.

Please understand that my idea will not take anything away from your "normal users". That appears to worry you, but it's based on a misunderstanding.

bomilkar, 2022-08-11:
> The data partition should be mounted `noexec`, and I don't like the idea of putting executables on the data partition

@MaxKellermann can you explain why? Is it just to keep things seperate or is there a deeper reason?

MaxKellermann, 2022-08-11:
> > The data partition should be mounted `noexec`, and I don't like the idea of putting executables on the data partition
> 
> @MaxKellermann can you explain why? Is it just to keep things seperate or is there a deeper reason?

The former. Since this takes away (useless) features, nothing really improves by adding "noexec".

Being strict helps reveal mistakes/misdesigns as early as possible. That includes things like making the root/system partition read-only, and spraying mount flags like "noexec", "nosuid", "nodev" on all other mounts.

This is a question of taste. Many people don't like being strict; those are the people who don't like compiler warnings, static analyzers or any proactive security measure. Being strict forces them to come up with "different" (better?) designs, and being forced to do things differently makes them uncomfortable. I think those people are wrong, but that's just my taste.

Of course, it can be useful to allow code execution from the data partition. On my Kobo, I often drop executables (e.g. strace, gdbserver) on the data partition and execute them from there. For that, I may need to remount with "exec", but that's okay, this is a very rare corner case for expert hackers. For "regular" users, having "noexec" by default is better.