Openvario/meta-openvario #187: Image 22005: Sound of variod not contolable from XCSoar

eku60, 2022-01-16:
Hello to the group,

I used image 22005 from openvario.org and two images 22002 which were created by myself. With none of the images (CH07) sound volume or sound mute could be controlled. Also switching Vario/STF inside XCSoar was not possible.

Any ideas?

Regards
eku60

iglesiasmarianod, 2022-01-18:
Hi eku60, we had the same problem when OS changed from Armstrong to Warrior, it was a socket disconnection issue.
Can you please post an XCSoar log from your device?

August2111, 2022-01-19:
Hello eku60,
with the newest self build image from hardknott (commit 73fbee0123) it is working as expected - with the former one (image 22005) I had the same problem...
From my point of view it is solved!

eku60, 2022-01-23:
Hello to the group,

sorry for the delay, but I had a harddisk-crash and to perform a complete new installation of 21.10 (Ubuntu). I'm sure the HD died slowly, what was responsible for my problems.

Then installed repo an docker on the brand new machine and, after a while i had new ov-image: 22023, size 1.097.728 KB.
Very proud, and then got the same mistakes kedder and the others above....

XCSoar not starting, Shell leads to the main-mainu written in an larger font.

Screen is showing...
XCSoar config file adaption
Mount config dir ...
Doing DisplayOrientation ...
Config string found
Rotation
Replacing ... XCSOAR_ROTATION=
Unmount config dir ...
Exiting ...
Using DRI device /dev/dri/card0 (use environment variable ...DRI..CE to override)

and the main menu comes back, all the time.

Regards 
eku60

August2111, 2022-01-23:
The first time I checked out the repository I had the same effect. After that I went back to #PR172 (commit 06863c83f4) in the repo meta-openvario to find out from which PR/commit this error occurs (going the PRs step by step) - with the result that in the end it always runs for me and starts properly ;-) - up to the last  #PR191 (commit 73fbee012384)...
It is of course very difficult to fathom what is missing in a 'virgin' setup...
The last 2 PRs from Max I didn't check up to now, but I think it will work properly too!

eku60, 2022-01-24:
Hello August2111,

the settings in repo from yesterday (24 Jan. 2022) in the morning produces an image which will not start XCSoar. 
It is useless for me and I don't want to make tests with going back to whatever PRs for getting a running system. This costs a lot of time, and one cannot say what is missing then. I don't know whether or not the last 2 PRs of him work properly.
Something, however, does not work properly so far and needs repair.
I'm sure, there will be a solution. 
Good to know, that ist is not only my problem, so my "bitbake" should not have been the problem. :)

Regards
eku60

August2111, 2022-01-24:
Hello eku60,
yes, you are right. In the last 14 days there are a lot of investgations started to make the OpenVario and the build process better in the future...
Unfortunately it was forgotten to keep a fixed branch with a current funcional level for users outside the developer team - to create an own image based on a working stand! But unfortunately I have no influence to reorganice something, so I will ask to do that from the maintainers!

linuxianer99, 2026-03-18:
Please retest with the new upcoming release.