Openvario/meta-openvario #200: Logo SVG file

MaxKellermann, 2022-01-22:
The logo should be a SVG file which will be converted to raster bitmap during the build. This way, we don't need many copies of the same BMP files, and generate those we want in any resolution for all display sizes.
Does somebody have a SVG file of it?

mihu-ov, 2022-01-23:
[image]

MaxKellermann, 2022-01-24:
Thanks, who is the copyright holder and what is the license?

MaxKellermann, 2022-01-24:
This is .... not a SVG file. Technically, it is, but it is just a SVG container with an embedded PNG raster file, encoded in base64. No vector graphics at all, not scalable.

mihu-ov, 2022-01-24:
I will check with Guido.

mihu-ov, 2022-01-31:
Just to let you know Guido is working on it.

MacflySF27, 2022-02-05:
Hi! My daughter made new “refurbished” logos.
The favourite one with a color gradient -and one with two colours (just in case…)

…hope this works!?

all the best - Guido

[image]
[image]

lordfolken, 2022-02-05:
Hello guide, this looks really good and is in the proper format.
 I have one request however: The logo should be visible also on a black/dark background, would some shading be possible?
We should also create an AUTHORS file and credit her for her work.

lordfolken, 2022-02-05:
#258

MacflySF27, 2022-02-05:
Hello!
I just talked to my daughter - and she said that shading or vignetting ist not the way to go - "industry std." is a second logo where the dark colours are inverted.
What do you think about this idea?

lordfolken, 2022-02-06:
I'm fine with an inverted logo aswell :)

lordfolken, 2022-02-06:
@linuxianer99 where in the source tree should we add this? We have a Data/graphics/ in XCSoar, perhaps a similar approach?

MaxKellermann, 2022-02-06:
I'll take care of integrating it, and auto-generate the BMP files.

MaxKellermann, 2022-02-06:
@MacflySF27 for the git commit adding the SVG file, what shall I enter as its author?
What is the license? (You could say choose between GPLv3 and CC-BY-SA at will, so both licenses are covered, and the logo could be used by both OpenVario documentation and the software)

MacflySF27, 2022-02-07:
Hi!
Finally my daughter did all the work - her name is 
Andrea Probst and you can link her homepage
www.andpro.work.
License - sorry no idea - your choice.

Here is the inverted logo - i think it looks great on a black background:
[image]

here is a screenshot with black background:
[image]

closed 2022-02-14 by linuxianer99