Openvario/meta-openvario #201: psplash missing --angle switch

lordfolken, 2022-01-23:
psplash reads the rotation configuration either from the /etc/rotation config file (angle in degrees) or from the commandline for example: ```--angle 90```

lordfolken, 2022-01-23:
hmm the /etc/rotation doesn't seem to work.

lordfolken, 2022-01-24:
So i had a look at the psplash-init script and modifed it:
```bash
#!/bin/sh 
### BEGIN INIT INFO
# Provides:             psplash
# Required-Start:
# Required-Stop:
# Default-Start:        S
# Default-Stop:
### END INIT INFO

. /etc/default/rcS
export PSPLASH_FIFO_DIR

if [ ! -e /dev/fb0 ]; then
    echo "Framebuffer /dev/fb0 not detected"
    echo "Boot splashscreen disabled"
    exit 0;
fi

read cmdline < /proc/cmdline
for x in $cmdline; do
        case $x in
        psplash=false)
		echo "boot splashscreen disabled" 
		exit 0;
                ;;
        esac
done

rotation=0
read cmdline < /proc/cmdline
for x in $cmdline; do 
       case $x in 
         fbcon=rotate:0)
             rotation=0
             ;;
         fbcon=rotate:3)
             rotation=90
             ;;
         fbcon=rotate:2)
             rotation=180
             ;;
         fbcon=rotate:1)
             rotation=270
             ;;
       esac
done

[ -d $PSPLASH_FIFO_DIR ] || mkdir -p $PSPLASH_FIFO_DIR
if ! mountpoint -q $PSPLASH_FIFO_DIR; then
	mount tmpfs -t tmpfs $PSPLASH_FIFO_DIR -o,size=40k
fi

/usr/bin/psplash --angle $rotation 
```
But now it seems that the psplash uses the systemd.service file.

lordfolken, 2022-01-26:
Ok the above approach was wrong, as we are using systemd service of psplash now. 

I have a working solution: 

in the psplash.service file i added the --angle " ${ROTATION}" switch. 
and an EnvironemntFile=/etc/rotation
the environmentfile containes: ROTATION=0

I extended ovmenu-ng to write the /etc/rotation file according to the selection. 

What i don't like is to have the setting in two places, once for the framebuffer and once for the bootsplash screen. 

Alas the uboot kernel parameteres need fbcon=rotate=3, wheras --angle needs degrees. i.e. 90.

My current challenge is to add this to bitbake.
The /etc/rotation file needs to be installed, or else the splash screen will not show on the first boot.

linuxianer99, 2022-02-04:
@lordfolken Think this is closed with #219  right ?

lordfolken, 2022-02-04:
correct.

closed 2022-02-04 by lordfolken

linuxianer99, 2022-02-04:
@lordfolken Thanks :-)