Openvario/meta-openvario #202: exit to shell, restarts menu

lordfolken, 2022-01-23:
When trying to use ovmenu's exit to shell function, it actually restarts ovmenu. 
The only way to get to a shell is to alt-f2 and login as 'root'

Scumi, 2022-01-23:
This is the new behaviour according to #190

lordfolken, 2022-01-23:
I.c. I agree to remove this function from ovmenu.
In general an "normal" user shouldn't need to change to the shell.

eku60, 2022-01-24:
Hello,

as long as things like backups of the .xcsoar directory do not work as they should, an easy way to go to the shell would be very helpful (remember the discussion on $HOME - usage).
The "normal" user will always let the timer run down (3 seconds!) and let start XCSoar. But what will the "normal" user do to connect to wifi if he can't call things like connmanctl?
How can the polar_a(b,c,w) values be set when not beeing able to edit variod.conf or sensord.conf? How to deal with, if one would like to change the settings for quiet-values and so on?
Shell and Nano makes this a lot easier. Vi? Vi is a tool one must like very well. Not for usual users.

So before such tools are eliminated, much of that stuff has to be included in a "settings" environment.

With an image 22023 (yesterday), it's not possible to start XCSoar.

Just my two cent. Originally I wanted to be a "normal" user...
Regards
eku60

lordfolken, 2022-01-24:
@eku60 don't worry logging in to the shell you can still do with alt-f2 and login via the tty. (and that functionality will remain, as we are all hackers.)

Its just that the function advertising this doesn't presently work in the ovmenu, as its now executed via systemd as a system service. So exiting it will just restart the "ovmenu" service and not drop to the encompassing shell.

closed 2022-01-25 by linuxianer99