Openvario/meta-openvario #203: ov57-lvds Display doesn't turn off when selecting poweroff

lordfolken, 2022-01-23:
Build: https://github.com/Openvario/meta-openvario/commit/1e1c6a9a596e5b49e99fc3e24bb41e2e4581e8bc
Machine: ov57-lvds (DS1)

The display on my ov57-lvds doesn't turn off when powering off. It fades to a white washed out screen and the backlight stays on. 
I then have to pull the fuse (both fuse/switch), to effectively power-off the device.

iglesiasmarianod, 2022-01-23:
Same behaviour here. Perhaps #103 might shed some light.
@kedder patched it for Warrior, I don't know if there's a full fix yet.

MaxKellermann, 2022-01-24:
The patch is still there, but appears to have no effect now. It's back to random behavior.

MaxKellermann, 2022-01-24:
Oh, look at this: https://github.com/torvalds/linux/commit/6eefb79d6f5bc4086bd02c76f1072dd4a8d9d9f6
This was also merged in 5.10.4, but Yocto-Hardknott is still at 5.10.2. (I hate using outdated kernels.)

lordfolken, 2022-01-24:
I upgraded it manually to 5.10.4, but no change. powering off from ovmenu or poweroff from the shell.

MaxKellermann, 2022-01-24:
That's what I meant when I wrote "appears to have no effect now". Something else broke it.
Unfortunately, there are no relevant fixes for this driver in newer kernels.

lordfolken, 2022-01-24:
For completion sake i also tried 5.10.93. Still no change.

MaxKellermann, 2022-01-25:
Looking at the PWM driver patch, I believe it never had any effect. This is a timing problem. I think the driver miscalculates the `next_period`, and submits the "disable" command too early.

kedder, 2022-01-25:
I can assure you it did fix an issue at that time. Not hard to verify by loading the old image. It's something else now, but with seemingly similar symptoms.

MaxKellermann, 2022-01-25:
I have found the real problem - it's a rounding error which has always been there, which leads to miscalculation of `next_period`, so my initial theory is correct.

The patch removing the `else` branch really does not solve the problem. It may change the timings in random ways that appear to make the problem go away, but that's just that: random coincidence.

kedder, 2022-01-25:
How random though? I didn't experience any randomness after the fix. Before the fix the "randomness" was clearly corellated with the brightness, as described here: https://github.com/Openvario/meta-openvario/issues/103#issuecomment-681132576.

MaxKellermann, 2022-01-25:
The correlation with the brightness is truly mysterious. I cannot see any reason for that in the source code. Same goes for why this `else`-removing patch changes the probabilities. When I added some debugging code, the problem also just went away. If you're dealing with timing problems, nothing is deterministic.

The bug is this line:
```
	sun4i_pwm->next_period[pwm->hwpwm] = jiffies +
		nsecs_to_jiffies(cstate.period + 1000);
```
So this converts the `period` from nanoseconds to "jiffies", but the result is always zero (at `CONFIG_HZ=100`, because the period is 5ms). That is, the driver will never sleep, resulting in two writes to the register, only delayed by the `spin_lock()` call - and here's where the randomness is injected which makes this sometimes just work and sometimes not.

kedder, 2022-01-25:
If I remember correctly, it correllated with brightness because the driver was not setting the pwm line to "off" state, it just merely turned off the clock, so the line was left in whatever state it was at that moment - either full on or full off. I don't remember all the details by now, but the solution was to set the PWM duty cycle to 0 before turning off the clocks.

closed 2022-01-25 by linuxianer99