Openvario/meta-openvario #204: Reboot loop on latest image

kedder, 2022-01-23:
Image, built from the latest `hardknott` (1e1c6a9a596e5b49e99fc3e24bb41e2e4581e8bc) does not boot: the splash screen shows up briefly, then couple of seconds of black screen and then the device reboots.

Will try to find more clues on what is going on.

kedder, 2022-01-23:
Relevant kernel log snippet below. It looks like the image is indeed only 6.1M in size and only contains root partition. Th image I'm looking at is `build/tmp/deploy/images/openvario-7-CH070/OpenVario-linux-openvario-image-glibc-ipk-22023-openvario-7-CH070.rootfs.img.gz`. 

@MaxKellermann do you have a clue what's going on here?

```
[    2.109180] List of all partitions:
[    2.112695] b300         3952640 mmcblk0 
[    2.112700]  driver: mmcblk
[    2.114455] usb 1-1: new high-speed USB device number 2 using ehci-platform
[    2.116772]   b301           40960 mmcblk0p1 a7a26646-01
[    2.126522] 
[    2.133340]   b302         1052672 mmcblk0p2 a7a26646-02
[    2.133343] 
[    2.140187] No filesystem could mount root, tried: 
[    2.140192]  ext3
[    2.145089]  ext2
[    2.147014]  ext4
[    2.148936]  vfat
[    2.150857]  msdos
[    2.152784] 
[    2.156304] Kernel panic - not syncing: VFS: Unable to mount root fs on unknown-block(179,2)
[    2.164735] CPU: 0 PID: 1 Comm: swapper/0 Not tainted 5.10.2 #1
[    2.170645] Hardware name: Allwinner sun7i (A20) Family
[    2.175892] [<c010c78c>] (unwind_backtrace) from [<c01093a4>] (show_stack+0x10/0x14)
[    2.183633] [<c01093a4>] (show_stack) from [<c06af93c>] (dump_stack+0x94/0xa8)
[    2.190853] [<c06af93c>] (dump_stack) from [<c06ad3c0>] (panic+0xf8/0x2e4)
[    2.197727] [<c06ad3c0>] (panic) from [<c09013fc>] (mount_root+0x0/0x78)
[    2.204425] [<c09013fc>] (mount_root) from [<c1002628>] (0xc1002628)
[    2.210782] CPU1: stopping
[    2.213494] CPU: 1 PID: 0 Comm: swapper/1 Not tainted 5.10.2 #1
[    2.219404] Hardware name: Allwinner sun7i (A20) Family
[    2.224632] [<c010c78c>] (unwind_backtrace) from [<c01093a4>] (show_stack+0x10/0x14)
[    2.232370] [<c01093a4>] (show_stack) from [<c06af93c>] (dump_stack+0x94/0xa8)
[    2.239588] [<c06af93c>] (dump_stack) from [<c010b344>] (do_handle_IPI+0xf4/0x12c)
[    2.247152] [<c010b344>] (do_handle_IPI) from [<c010b394>] (ipi_handler+0x18/0x20)
[    2.254717] [<c010b394>] (ipi_handler) from [<c016371c>] (__handle_domain_irq+0x7c/0xd0)
[    2.262805] [<c016371c>] (__handle_domain_irq) from [<c038532c>] (gic_handle_irq+0x80/0x94)
[    2.271151] [<c038532c>] (gic_handle_irq) from [<c0100b0c>] (__irq_svc+0x6c/0x90)
[    2.278623] Exception stack(0xc0c83f58 to 0xc0c83fa0)
[    2.283669] 3f40:                                                       00000000 00002160
[    2.291839] 3f60: ef7ac844 c0112760 c0c82000 00000001 c0a04ed0 c0a04f0c 00000000 c0956280
[    2.300008] 3f80: 00000000 c0c83fb0 00000001 c0c83fa8 c0106f08 c0106f0c 60000013 ffffffff
[    2.308182] [<c0100b0c>] (__irq_svc) from [<c0106f0c>] (arch_cpu_idle+0x38/0x3c)
[    2.315576] [<c0106f0c>] (arch_cpu_idle) from [<c0143830>] (do_idle+0x210/0x2bc)
[    2.322968] [<c0143830>] (do_idle) from [<c0143b78>] (cpu_startup_entry+0x18/0x1c)
[    2.330530] [<c0143b78>] (cpu_startup_entry) from [<4010152c>] (0x4010152c)
[    2.337491] Rebooting in 10 seconds..
```

lordfolken, 2022-01-23:
You may have run out of diskspace on your build machine.

kedder, 2022-01-23:
> You may have run out of diskspace on your build machine. I'm running that commit on my ov.

That's definitely not the case there are still 25GB of free space and build completed successfully.

kedder, 2022-01-23:
Maybe that's just me? I'll try a completely fresh build from scratch overnight.

lordfolken, 2022-01-23:
well i was able to build the image. but XCSoar refuses to run 
with ```drmModeGetResourcesFailed()```

lordfolken, 2022-01-23:
looks like i have now 2 /dev/drm/dri/card0 and card1 setting DRI_DEVICE=/dev/drm/dri/card1 and xcsoar works.

kedder, 2022-01-23:
Hm, I believe there was a special configuration to prevent a random device name assignment for lima and dri devices (both registering as `/dev/dr/dri/card*`). Need to look it up again...

lordfolken, 2022-01-23:
also power and cpu usage  (30-40%) is much higher.. i have now 0.85 amps compared to 0.6 before (including xcvario)

kedder, 2022-01-23:
ok, after all that was just my local issue. I have an image that boots, but I can confirm the same issue as @lordfolken reports (`drmModeGetResourcesFailed()`). I'll close this issue and reopen another one.

closed 2022-01-23 by kedder

kedder, 2022-01-23:
@lordfolken for the power usage - make sure you are comparing images with the same screen brightness level (by default new install starts on maximum brightness).  If both images are on the same screen brightness, please file a separate issue.