Openvario/meta-openvario #207: XCSoar doesn't start

kedder, 2022-01-23:
On the latest `hardknott` image  (1e1c6a9), XCSoar fails to start. The device shows this message on screen:

```
Using DRI device /dev/dri/card0 (use environment variable DRI_DEVICE to override)
drmModeGetResources() failed
```

After few seconds the main menu appears.

The issue is reported in #204 by @lordfolken. It is possible to run XCSoar from command line with `DRI_DEVICE=/dev/drm/dri/card1` environment variable.

MaxKellermann, 2022-01-24:
Fixed in xcsoar-testing (-> 7.23)

lordfolken, 2022-01-27:
What do we do here? Do we wait until 7.23 is released? This bug is still in the current master.

MaxKellermann, 2022-01-27:
Yes. I think the next XCSoar release will be long before the next OpenVario release.

lordfolken, 2022-01-27:
thren at least the bug report should stay open, as there is still a change needed in meta-openvario.

linuxianer99, 2022-01-27:
ok ... you are right ... sorry ...

MaxKellermann, 2022-02-10:
https://github.com/Openvario/meta-openvario/pull/268

closed 2022-02-11 by linuxianer99