Openvario/meta-openvario #208: ovmenu-ng ignores touch screen input

MaxKellermann, 2022-01-24:
After booting my OpenVario (thanks @mihu-ov), I tried to cancel the timer which launches XCSoar to enter the ovmenu, but it wasn't possible with the touch screen. I can use XCSoar just fine with just the touch screen, but the ovmenu simply ignores any touch input.

mihu-ov, 2022-01-24:
We never had touchscreen input for ovmenu. OV started with touch input but when stick remotes and rotary encoders became available the touchscreen became less attractives. Most OVs probably don´t have a touchscreen installed as it creates additional reflective layers and reduces sunlight readability but some people still use it.
The electronics of an early version of the stick remote is in the box but a keyboard is probably more useful for testing. ESC (the X on newer stick remotes) stops the timer and enters ovmenu.

lordfolken, 2022-01-27:
Still a mouse emulation should be possible, Maybe good old [gpm](https://wiki.archlinux.org/title/General_purpose_mouse) ?