Openvario/meta-openvario #217: XCSoar unstable when downloading

lordfolken, 2022-01-25:
Platform: ov57-lvds
XCSoar: 7.22

Several times, i had it now that when i try to download a file via the file manager, and select it (for example the waypoints), that xcsoar justs exists. The log unfortunately doesn't say any lines beyond startup. 

I'll add this here as a placeholder. It will be updated as soon as i gain information, or steps to reproduce.

lordfolken, 2022-01-28:
So the most reliable way to crash it:
 switch to alt-f2
 start xcsoar -simulator
 config->SiteFiles->Maps-Download->ALPS_highres.xcm

The download then completes according to the progressbar. 
XCSoar then either exits, or the whole OS crashes.

MaxKellermann, 2022-01-28:
Unfortunately, I still cannot reproduce. Tried like 10 times, and everything works as expected.

linuxianer99, 2022-02-04:
@lordfolken Is this still an issue ??

lordfolken, 2022-02-14:
i cannot reproduce anymore. so i close.

lordfolken, 2022-02-17:
This is still very much the case.

Downloading a large file, xcsoar crashes on openvario-image-testing

kedder, 2022-02-17:
@lordfolken is your OV system time up-to-date when you try to download something. Is it trying to download things using https?

kedder, 2022-02-17:
Actually, if download (apparently) completes, this would not be certificate verification issue. Do you have enough free space on disk?

kedder, 2022-02-17:
We are trying to narrow down the issue and make it reproducible.

MaxKellermann, 2022-02-17:
OK then, try it your way.

lordfolken, 2022-02-17:
Space: fresh image.. 700+mb free
Timesync: if time isn´t synced and the certificate is invalid, there is a popup "certificate not yet valid" when trying to download the file.

lordfolken, 2022-02-17:
Btw, where does xcsoar download the file to? At least during the download, i don't see the file in .xcsoar/

lordfolken, 2022-02-17:
So it completes the download of ALPS_highres and then segfaults before returning to the settings menu with newly downloaded file.

MaxKellermann, 2022-02-17:
> Btw, where does xcsoar download the file to? At least during the download, i don't see the file in .xcsoar/

XCSoar uses `O_TMPFILE`, and the file is linked to the filesystem only after the download is finished.

bomilkar, 2022-09-25:
I tried it and agree: https://github.com/XCSoar/XCSoar/issues/1053 addresses this issue. (As of 22266).

closed 2022-09-26 by lordfolken