Openvario/meta-openvario #221: Wrong bblayers.conf in out-of-tree build

MaxKellermann, 2022-01-27:
After the repository was restructured (https://github.com/Openvario/meta-openvario/issues/214), I tried starting a fresh build. But I didn't want to have the build files in my home partition (which is small and encrypted), so I tried an out-of-tree build:
```
~/git/meta-openvario$ source openembedded-core/oe-init-build-env /srv/openvario
/srv/openvario$ bitbake openvario-image-testing
[...]
ERROR: Nothing PROVIDES 'openvario-image-testing'
```
Turns out `conf/bblayers.conf` contains *only* `openembedded-core`.

linuxianer99, 2022-01-29:
I tried to fix this ..
Putting a .templateconf file in meta-openvario, but doesn't work.
Do you have an idea, how to fix it ?

linuxianer99, 2022-02-01:
@MaxKellermann   Do you have an idea, how to fix this ?