Openvario/meta-openvario #226: usb stick is mounted to wrong directory

lordfolken, 2022-01-27:
The stick is mounted directly on /usb. All the scripts expect "/usb/usbstick" as their primary path. 

The user_guide appendix b specifies the following layout:

```bash
/
   openvario/
      maps/
      repo/
      download/
          xcsoar/
      upload/
          xcsoar/
      image/
```
The download/upload scripts then recreate this directory structure below usbstick/

MaxKellermann, 2022-01-28:
Regression from commit 01a2341385687e83327b401fe1bdd723650422c8
Apparently, my interpretation of the autofs daemon config was wrong...
Why are we using a non-standard top-level path like `/usb/usbstick`, instead of, say, `/mnt/X` or `/media/X`?

lordfolken, 2022-01-28:
The /media/<user>/<device label> is typically managed by a server/client combo called udisks2, that triggers on dbus messages upon changes in storage device avalability.

Typically autofs starts with a root of the type, like /nfs or /smb, or as in this case /usb. 
The reason you want a separate toplevel directory for autofs, is that any access to it triggers a mount (attempt). So with this directory, intent is clearer.

lordfolken, 2022-01-28:
SUSE has great docs (as always) on [autofs](https://documentation.suse.com/sles/15-SP3/html/SLES-all/cha-autofs.html).

MaxKellermann, 2022-01-28:
> The `/media/user/device label` is typically managed by a server/client combo called udisks2, that triggers on dbus messages upon changes in storage device availability.

I know udisks2 (I have written [lots](https://github.com/MusicPlayerDaemon/MPD/blob/master/src/storage/plugins/UdisksStorage.cxx) [of code](https://github.com/MusicPlayerDaemon/MPD/blob/master/src/neighbor/plugins/UdisksNeighborPlugin.cxx) [interacting with it](https://github.com/MusicPlayerDaemon/MPD/blob/master/src/lib/dbus/UDisks2.cxx)), and I know `pmount`, which also uses `/media/`. Because that's what the [FHS standard says where removable media shall be mounted](https://refspecs.linuxfoundation.org/FHS_3.0/fhs/ch03s11.html).

> Typically autofs starts with a root of the type, like /nfs or /smb, or as in this case /usb. The reason you want a separate top-level directory for autofs, is that any access to it triggers a mount (attempt). So with this directory, intent is clearer.

With mount points below `/mnt` or `/media`, you would have the same effect, and you wouldn't be violating FHS.

> SUSE has great docs (as always) on [autofs](https://documentation.suse.com/sles/15-SP3/html/SLES-all/cha-autofs.html).

We're not using autofs anymore, because systemd can do the same, just better.

What I don't understand is why we use an auto-mounter at all. I think mounting and unmounting should be explicit, to avoid data loss and freezing processes. Making this dynamic would also allow more than one USB stick, or would allow using a SATA disk - having a SATA disk currently breaks everything, because `/dev/sda1` is hard-coded.
The current OpenVario setup is very fragile, and I don't like that.

lordfolken, 2022-01-28:
I don't really disagree on any particular point. I also think we should open a design discussion on storage layout and handling. 
 * separate root / home (userdata) partitions
 * separate rescue partition
 * stop running as root

The purpose of this bug report is just to say that it doesn't work like it was designed due to an error. 

My goal is that master is usable and that we can incrementally transform it from there.

linuxianer99, 2022-01-29:
Feel free to start a new topic in "Discussions" !!

lordfolken, 2022-02-03:
I will close this issue, as the original problem is resolved.

closed 2022-02-03 by lordfolken