Openvario/meta-openvario #229: add rsync to image

lordfolken, 2022-01-28:
Please add rsync to the image:
 * copying from / to usb stick would be more efficient with the delta algorithm
 * --progress (is a progress bar)
 * --delete (would cleanup directories of no longer present files at the source)
 * even works via ssh

MaxKellermann, 2022-01-28:
Works only if the RTC on the Cubieboard is reliable (or else data loss). And without `-u`, this isn't efficient.

lordfolken, 2022-01-28:
True the systemtime (or lack thereof) has to be considered:
```dmesg
[    1.190385] sunxi-rtc 1c20d00.rtc: setting system clock to 1970-01-01T00:00:13 UTC (13)
```
there is -I for ignoring the times. Checksumming is still better than writing, at least on my usb sticks read to write is around factor 10 faster than writing. Plus the other advantages still stand.

MaxKellermann, 2022-02-03:
btw. we shouldn't "add rsync to image"; the image should not have dependencies like that.
If ovmenu-ng-skripts one day wants to use rsync, it will have a RDEPENDS on rsync. As long as nobody uses rsync, it should not be installed.
So this issue should be a feature request "ovmenu-ng-skripts should use rsync instead of cp".

linuxianer99, 2022-02-03:
I agree with Max.

@lordfolken : Should we close this issue and open another one, as Max suggested ?

closed 2022-02-03 by lordfolken

kedder, 2022-02-03:
`rsync` is available for `opkg install` through my [PPA](https://github.com/kedder/openvario-ppa) if anyone is interested.