Openvario/meta-openvario #232: language at XCSoar is always English

Blaubart, 2022-01-29:
I'm not able to change language at XCSoar using the OpenVario menu.

bomilkar, 2022-02-02:
Make sure LANG is set before xcsoar starts.
For instance "LANG=de_DE.UTF-8".
This is a bad workaround but it may help pinning down the root cause.

MaxKellermann, 2022-02-02:
This is because `ovmenu-ng.sh` is bugged:
https://github.com/Openvario/meta-openvario/blob/42b5de934d3d95f76ca514c069cac90d01f1145b/meta-ov/recipes-apps/ovmenu-ng/files/ovmenu-ng.sh#L13
On my OpenVario, this expands to:
```
source /opt/conf/machine.conf /opt/conf/ov-xcsoar.conf /opt/conf/sensord.conf /opt/conf/variod.conf
```
The author expected this to read all `*.conf` files, but it does not.
Look at https://www.gnu.org/savannah-checkouts/gnu/bash/manual/bash.html#Bourne-Shell-Builtins
> . filename [arguments]

This will only read the first file (i.e. `machine.conf`) and pass the names of all other files as command-line arguments to it.

Fortunately, it did not execute all files ... because `sensord.conf` and `variod.conf` files are not shell scripts. Executing them spits some error messages at best.

closed 2022-02-02 by linuxianer99