Openvario/meta-openvario #235: ovmenu-recovery.sh redirects update-system.sh output to log file

MaxKellermann, 2022-01-31:
Here `update-system.sh` gets invoked, but its stdout gets redirected to a file:
https://github.com/Openvario/meta-openvario/blob/e662707c8d777d113f738cca5088dcf84f1be67b/meta-ov/recipes-apps/ovmenu-recovery/files/ovmenu-recovery.sh#L170-L171
This does not make any sense, because this is an interactive script:
https://github.com/Openvario/meta-openvario/blob/e662707c8d777d113f738cca5088dcf84f1be67b/meta-ov/recipes-apps/ovmenu-ng-skripts/files/update-system.sh#L16
This cannot work. But this is the only user of this script (and actually, its use is commented out).
What's the explanation for this strange piece of code?

linuxianer99, 2022-02-01:
> Here `update-system.sh` gets invoked, but its stdout gets redirected to a file:
> 
> https://github.com/Openvario/meta-openvario/blob/e662707c8d777d113f738cca5088dcf84f1be67b/meta-ov/recipes-apps/ovmenu-recovery/files/ovmenu-recovery.sh#L170-L171
> 

The tail.%$$ file is used as buffer for the next window in line 171. So the output of the update-system.sh is show in the tailbox in line 171.

> This does not make any sense, because this is an interactive script: [...]

I never used this script ... So I have no idea ;-)

MaxKellermann, 2022-02-01:
> The tail.%$$ file is used as buffer for the next window in line 171. So the output of the update-system.sh is show in the tailbox in line 171.

I can see that, but my point is that redirecting stdout breaks anything inside the called script which wants to interact with the user.

linuxianer99, 2022-02-01:
Hmm ... you are right ....
Strange, that nobody had this problem in the past ...

In general, we have to thing about the recovery concept of Openvario ... maybe it is also a good idea, to have the XCSoar-menu based approach also in the recovery ITB image ?

MaxKellermann, 2022-02-01:
> In general, we have to thing about the recovery concept of Openvario ... maybe it is also a good idea, to have the XCSoar-menu based approach also in the recovery ITB image ?

Yes, why not. Usually, I'd say a GUI toolkit is too bloated, but a monolithic executable with XCSoar's GUI toolkit is smaller than just bash alone ...

linuxianer99, 2022-02-01:
> Yes, why not. Usually, I'd say a GUI toolkit is too bloated, but a monolithic executable with XCSoar's GUI toolkit is smaller than just bash alone ...

Yes! And it can be used via touch ;-)