Openvario/meta-openvario #251: New release and defined baseline

linuxianer99, 2022-02-04:
Hey all!

As we integrated lots of improvements the last week, i suggest to declare a Release Candidate ....

I my mind it makes sense to get a status, what is working now and where are some issues, which have to be fixed.

@August2111 @lordfolken Can you please make some basic tests, if the image can be used by "normal" users in flight, or if there are still some bugs to fix ??

@MaxKellermann @kedder @lordfolken @August2111 Do you agree in general with my point ??

If so, i would tag the current source as V3.0-RC1.
Side note: We have still to check, if the new version numbering breaks anything ...

kedder, 2022-02-04:
Good idea, let's do that!

kedder, 2022-02-04:
I think we'd need a new XCSoar release for #207 though, in order for users to do any testing.

linuxianer99, 2022-02-04:
I created a project 

https://github.com/orgs/Openvario/projects/1/views/1

lordfolken, 2022-02-04:
I would agree. 

We  could open a milestone in github, and start triaging stuff we see necessary for that release?
Also if we could use labels in issues... So we can separate with enhancement and bugs.

The current prs i would see: 
#245
#220 
Some bugs:
#230 
#225
#224 
#207 as @kedder mentioned.
and probably some others.

lordfolken, 2022-02-04:
We would also need to do some testing, with usb stick upgrades. So far i only imaged with dd, but didn't do a "field" upgrade.

August2111, 2022-02-04:
>  So far i only imaged with dd, but didn't do a "field" upgrade.
@lordfolken  What do you exactly mean with "field upgrade"?

lordfolken, 2022-02-04:
pilot copying the new image to the usb stick and upgrade via recovery. Not removing the sdcard.

linuxianer99, 2022-02-04:
> I would agree.
>
> We could open a milestone in github, and start triaging stuff we see necessary for that release? Also if we could use labels in issues... So we can separate with enhancement and [...]

Added this issues to the project!

> We would also need to do some testing, with usb stick upgrades. So far i only imaged with dd, but didn't do a "field" upgrade.

I fully agree.

How has which hardware for testing ??
I have just 4,3 RGB and CH070

lordfolken, 2022-02-04:
57lvds

August2111, 2022-02-04:
What I really like is version information at XCSoar with NEWS.txt and VERSION.txt.

- With a living NEWS.txt you have a good overview what happens since the last release - otherwise it is very difficult to remember...
- With Version.txt you can link every time your version number to the image

lordfolken, 2022-02-04:
A prominent version number that is being displayed from a single source of truth. 
(For example in menu title bar, or mentioned in the splash screen)

What we could do better in xcsoar: is display those changes when you startup.

August2111, 2022-02-04:
> How has which hardware for testing ??

57-lvds without vario board
AM070-DS2 without vario board

From time to time I have acces to both types of device with vario in our club...

lordfolken, 2022-02-04:
Maybe we should device a testing plan?

MaxKellermann, 2022-02-04:
> I think we'd need a new XCSoar release for #207 though, in order for users to do any testing.

Unfortunately, there was not enough time this week - but I promise 7.23 will be ready next week!

linuxianer99, 2022-02-04:
> > I think we'd need a new XCSoar release for #207 though, in order for users to do any testing.
> 
> Unfortunately, there was not enough time this week - but I promise 7.23 will be ready next week!

No problem :-) We are all doing this in our spare time :-)