Openvario/meta-openvario #252: shell environment dosen't work correctly

Blaubart, 2022-02-04:
when I go to the shell environment, I can issue commands, but no longer see any output. The auto-completion with the tab key no longer works either.

MaxKellermann, 2022-02-04:
Could be caused by https://github.com/Openvario/meta-openvario/commit/a09ef7bade88ff549ccf6cb793d6673e0c3ad0a4 - please check if it works after removing that line (requires reboot).

MaxKellermann, 2022-02-04:
Also try redirecting stderr here (without reverting the systemd service change I cited above):
https://github.com/Openvario/meta-openvario/blob/2a69ab2b144e708f04f232bc9340033c34467bf7/meta-ov/recipes-apps/ovmenu-ng/files/ovmenu-ng.sh#L356-L362
Add to the bash line (and maybe to the other, fallback shell calls):
```
  /bin/bash --login 2>&1
```

Blaubart, 2022-02-04:
I removed the line:
StandardError=journal

and it works again

Should I check that, too?
"Also try redirecting stderr here (without reverting the systemd service change I cited above):"

MaxKellermann, 2022-02-04:
Yes, please. We know what caused the regression, and the second test would be a solution.

closed 2022-02-04 by linuxianer99

Blaubart, 2022-02-04:
I removed StandardError=journal and I changed at "meta-openvario/meta-ov/recipes-apps/ovmenu-ng/files/ovmenu-ng.sh" "/bin/bash --login"  to "/bin/bash --login 2>&1"

It still works