Openvario/meta-openvario #256: rotation is now bonkers

lordfolken, 2022-02-05:
Setting the rotation in ovmenu sets the rotation correct at that instant. If you reboot however both psplash and fb are rotated differently.

lordfolken, 2022-02-05:
its also only rotates the first console, not alt-f2.

lordfolken, 2022-02-05:
root@openvario-57-lvds:~# cat 3.prereboot 3.postreboot 
rotation=3
console=ttyS0,115200 root=/dev/mmcblk0p2 rootwait panic=10 lima.job_hang_limit=10 fbcon=rotate:2 vconsole.font=default8x16
3
rotation=3
console=ttyS0,115200 root=/dev/mmcblk0p2 rootwait panic=10 lima.job_hang_limit=10 fbcon=rotate:1 vconsole.font=default8x16
1```bash

lordfolken, 2022-02-06:
so basically when rotation=1 console should 3. and vica versa.

MaxKellermann, 2022-02-06:
Yes, I forgot to apply this translation:
https://github.com/Openvario/meta-openvario/blob/5f85d706c1744835db145129acc82c53a0bbbd61/meta-ov/recipes-bsp/u-boot/files/bootenv.ini#L37-L41
Can somebody explain to me why this translation exists? Why does OpenVario redefine the meaning of these rotation parameters, why are we not simply using the kernel's definition? This only complicates things, but does not appear to make sense. @linuxianer99, do you know?

lordfolken, 2022-02-06:
I found this now also after searching. I'm preparing a PR at the moment for this.

lordfolken, 2022-02-06:
I still have a problem with psplash. The /sys/class/graphics/fbcon/rotate returns 0 at boot, even if the consoles et all are rotated by the kernel argument.

lordfolken, 2022-02-06:
I suspect currently this kernel option:
CONFIG_FRAMEBUFFER_CONSOLE_DEFERRED_TAKEOVER=n

lordfolken, 2022-02-06:
setting  CONFIG_FRAMEBUFFER_CONSOLE_DEFERRED_TAKEOVER to n fixes the problem with psplash. (but displays penguins.)

MaxKellermann, 2022-02-06:
Indeed a surprising artefact - but why is it a problem?

linuxianer99, 2022-02-06:
> Yes, I forgot to apply this translation:
>
> https://github.com/Openvario/meta-openvario/blob/5f85d706c1744835db145129acc82c53a0bbbd61/meta-ov/recipes-bsp/u-boot/files/bootenv.ini#L37-L41
>
>
> Can [...]

@MaxKellermann 
This was a looooong time ago ;-) Maybe there was another rotation logic between Framebuffer and XCSoar config ? As XCSoar has auto-detection and we have now the Lima module, we can try to remove ?

lordfolken, 2022-02-06:
> Indeed a surprising artefact - but why is it a problem?

Googling for this option gives the following explanation:
"Linux 4.19 (to be released 2018-10-21) introduces deferred takeover of the framebuffer console, which is a new feature to avoid blanking the screen until console text needs to be printed. Instead, whatever vendor logo was left in the framebuffer from early boot is retained. By setting CONFIG_FRAMEBUFFER_CONSOLE_DEFERRED_TAKEOVER=y, a truly silent boot can be achieved in which the vendor logo is retained on screen until the login manager is ready to accept the user's password. This is described in further detail in the wiki:[ https://wiki.archlinux.org/index.php/Silent_boot#Retaining_the_vendor_logo_from_BIOS](https://wiki.archlinux.org/index.php/Silent_boot#Retaining_the_vendor_logo_from_BIOS)"

lordfolken, 2022-02-06:
So removing this would break the u-boot splash. (which currently doesn't display #248).
The question is whether we should read uboot parameter file, or the kernel cmdline as originally intended.?
@MaxKellermann

lordfolken, 2022-02-06:
An alternative would be to add a CONFIG_LOGO_CUSTOM=y and actually add the ov logo to the kernel.

lordfolken, 2022-02-07:
@MaxKellermann I suggest to revert:
https://github.com/Openvario/meta-openvario/commit/9abb7003489ba3d118a47a1cac60804002557d91 ?

With the CONFIG_FRAMEBUFFER_CONSOLE_DEFERRED_TAKEOVER kernel config parameter, /sys/class/graphics/fbcon/rotate is racy.

lordfolken, 2022-02-08:
https://github.com/Openvario/meta-openvario/pull/260

lordfolken, 2022-02-14:
This i believe is now solved with max's refactoring.