Openvario/meta-openvario #262: addtional kernel image in root

lordfolken, 2022-02-08:
When umount /boot you will find a kernel image /boot/
```bash
lrwxrwxrwx    1 root     root            13 Feb  8 21:28 uImage -> uImage-5.10.2
-rw-r--r--    1 root     root       5121552 Feb  8 21:28 uImage-5.10.2
```
we can save another 4.9MB.

MaxKellermann, 2022-02-09:
Stupid question - why does our image have a `/boot` partition? Right now, it appears to have no use, it only add redundancy and complexity.
@linuxianer99?

mihu-ov, 2022-02-09:
No technical answer, just some OV history: Linux on the CB2 started with non-mainline U-Boot and kernel provided by linux-sunxi.org. Even when U-Boot and kernel got mainlined we were still stuck on the old 3.4 kernel for MALI support until @kedder managed to make OV work with LIMA and mainline kernel. We probably have the /boot partition because that´s the way it was supposed to be at one stage of linux-sunxi development.

MaxKellermann, 2022-02-09:
OK, so we can easily remove the boot partition if we want.
I was just wondering if we should remove the kernel from `/boot` (on the root partition), or just eliminate the boot partition. The former is extra work, and the latter would not only save the time to do that, but also allow easier kernel updates - right now, the kernel cannot be updated with opkg; without a boot partition, it can.

linuxianer99, 2022-02-09:
> Stupid question - why does our image have a `/boot` partition? Right now, it appears to have no use, it only add redundancy and complexity. @linuxianer99?

Because the sdimg class in the sunxi-layer did this. There was no other reason. 
I am fine, to reduce complexity.

closed 2022-02-18 by linuxianer99