Openvario/meta-openvario #273: Vario does not work on the latest image

mihu-ov, 2022-02-16:
Reported by CazYokoyama https://forum.xcsoar.org/viewtopic.php?f=30&t=4008

Vario does not work on the latest image.
Post by CazYokoyama » Wed Feb 16, 2022 4:49 pm

The latest openvario image built from aa28fa60b229 has full up vario indication and no sound as shown in
http://www.caztech.com/glider/openvario ... 60b229.mp4
Is this expected?
The image is build by

source openembedded-core/oe-init-build-env .
export MACHINE=openvario-57-lvds
bitbake openvario-image

The same hardware with worrier image has up and down vario indication and sound.

CazYokoyama, 2022-02-16:
http://www.caztech.com/glider/openvario/openvario-aa28fa60b229.mp4

mihu-ov, 2022-02-16:
Thank you Caz, got the link wrong but it is corrected now. 
Can you check what goes wrong, from raw sensor data over I2C, or sensord or variod?

mihu-ov, 2022-02-17:
The vario gauge in Cazs video shows +99m/s so something is wrong, maybe sensord not able to read valid sensor data.
@linuxianer99 or @hpmax should be the experts on this.

MaxKellermann, 2022-02-17:
I found a way to work around the ALSA mixer problem; the way OpenVario chose to specify the initial mixer values is very fragile.
For other problems, please write a new bug report. (@CazYokoyama)

hpmax, 2022-02-17:
@MaxKellermann Could you elaborate?  I haven't tried a new release in a while but there was an issue with daemons just going to sleep occasionly that resulted in erratic timing and problems, particularly in audio generation, but also with reading the MS5611.

I've lost interest in the old sensorboard.   @DanD222 and I designed a new one and I'm working on a newer sensord to talk to it.

MaxKellermann, 2022-02-18:
Elaborate what?
How does a daemon go to sleep? Do you mean that a certain system call blocks? Which one?
(Side note: I'd like to get rid of sensord and integrate it fully into XCSoar.)

hpmax, 2022-02-18:
What question didn't I answer?

"How does a daemon go to sleep?" I don't know.  I suspect that Linux is simply not giving it a time slice to operate in for an extended time frame.  Why? Don't know.  Can I prove it?  Nope.

"Do I mean that a certain system call bocks?"  No, that's not what I meant although it's possible that's the explanation.  Sensord is accessing a few system calls on a regular basis... It's talking to the I2C bus.  It's talking to a socket to communicate with variod.  It's also reading the clock and putting itself to sleep.  Variod has the same system calls, except I2C, but it also talks to pulseaudio.  I suppose any of these calls (or none of these calls) could be blocking, but I was completely unsuccessful at determining why it was freezing.  All I know is it was far more prone to freezing when it was run as a daemon (from the /etc directory) than run from the command line.  Like I said, it's possible this problem was improved or fixed in a later release, based on a few comments that have been made but I don't have an explanation as to why.

hpmax, 2022-02-18:
Max, I have tried to provide accurate information.  If the information is inaccurate or incomplete it's likely I either don't know how to give you the information or I have forgotten the correct answer because it's been a year since I looked into this.  I gave up because no matter what I tried, I could not get it to work reliably despite efforts by numerous people on here.

Again, all I can describe is symptoms not root causes, and I believe sensord and variod are both suffering the same problem.  In the case of sensord I am trying to poll the ms5611 on a regular schedule and failing.  In the case of variod I am trying to keep an audio buffer from emptying and failing.  In both cases, this behavior is highly dependent on how the daemon was started.  If started manually from the command line it seems to work mostly okay -- if started through services it seems far more erratic.

If you look through the ms5611.c code in sensord you'll see there is a function called "sensor_wait()", which is used by pressure_measurement_handler() (in main.c).  Essentially at the beginning of pressure_measurement_handler() it calls sensor_wait() which uses nanosleep() to wake up when I want to take the next sample.  The wake-up time is calculated by taking the previous waking up time, adding a fixed time interval and subtracting the current time.  What I can tell you is that it sometimes leave sensor_wait() far later than it should.  Now, it's possible it's doing that because it arrived at sensor_wait() far later than it should and hence had a negative wait time, or it's possible it just nanosleep()'ed too long.  I believe the majority of time that sensord is running is being spent in sensor_wait() rather than actually doing anything.  At one time, I might have been able to tell you which one it was.  I don't remember any more if I ever knew.  It's possible it's documented somewhere.  It's worth pointing out that sensor_wait() was my idea to try to fix/improve the timing, I believe older versions of the code used delay() and they were prone to the same problem, so I don't think it's anything in sensor_wait.

MaxKellermann, 2022-02-18:
> If you look through the ms5611.c code in sensord you'll see there is a function called "sensor_wait()", which is used by pressure_measurement_handler() (in main.c).

I had a look at that function, but I have no idea what it does - it's undocumented... what does this parameter mean? What does its return value mean? Why is it floating point? What units are those parameters?
This looks like an extremely complicated way of doing what you explained. I tried to understand the commits you applied to this function, but I don't know what they mean; the changes to that function are hidden well inside huge commits that do lots of things at the same time. This makes your code even more obscure.

Oh, and it's buggy. Look at this:
```
1645201357.930753 clock_nanosleep_time64(CLOCK_REALTIME, 0, {tv_sec=0, tv_nsec=13181236906095259273}, 0xbec85b00) = 0 <0.001447>
```
This waits until the time stamp "0 seconds, 13181236906095259273 nanoseconds" has been reached. This is an illegal argument. This will do anything, but probably not what you expect it to do. It keeps the CPU pretty busy for nothing.

... and another question that beats me ... why the hell do you use `CLOCK_REALTIME` and not `CLOCK_MONOTONIC`? This may bring sensord into a defunct state whenever the real-time clock gets adjusted.

MaxKellermann, 2022-02-18:
> As it being buggy, you're going to have to elaborate, because I don't know what you are trying to point out.

The whole function does not make any sense. For example this:
```
	if (curtime.tv_nsec>1e9) {
		curtime.tv_sec=curtime.tv_nsec/1e9;
		curtime.tv_nsec-=curtime.tv_sec*1e9;
	} else {
		curtime.tv_sec=0;
		if (curtime.tv_nsec<0) curtime.tv_nsec=0;
	}
```

It `tv_nsec` is in range (i.e. less than 1 second), `tv_sec` gets reset to zero. This means sensord waits until the year 1970 has passed (the UNIX epoch, that's when tv_sec was zero). Yes, really. Year 1970.

This piece of code is so obviously wrong, that it confuses the hell out of me, because I can't explain why somebody would write it that way.

> As for failing... I'm not sure how I'm supposed to answer that. I am trying to poll the sensors at regular intervals. However, sometimes there are long delays between polling events, hence I am failing to poll at a regular interval.

How long is "long"?
How can I reproduce and measure that?

What is the practical problem of that?

hpmax, 2022-02-18:
How long is "long"? -- Typically long means on the order of 15-75 ms, when it's supposed to be 12.5.  I think I've seen a couple hundred ms in the worst case.  I believe even a few ms of error in timing jitter will produce a false vario reaction.

How can I reproduce and measure that? -- That's two different questions.  Measurement is done by using clock_gettime() and comparing it to what I was expecting.  Reproduce it is harder...  As I said, it has been alleged that the problem is fixed, however, the OV image I used (circa mid-last year) seemed to do it consistently.  Perhaps once every couple minutes of operation.

What is the practical problem of that? -- Again, the MS5611 reads incorrectly (I can't remember if it jumped high or low), from one sample to the next if there was a long delay.  It would come back to normal after a period of time, but that period of time could be a second or two.  While the change wasn't drastic, the rate of change was such that it would get picked up erroneously on the variometer.

Regarding the code...  I believe nanosleep() is a function to sleep for a certain amount of time, not until a certain time.  The function sets the sleep time to be prevtime + time (what was passed in) - curtime.  It then normalizes it by saying if the time is negative set it to 0, and if you're at more than a billion nanoseconds increment the seconds.

I'd like to point out that this function seems to work at least 99.99% of the time when run in the foreground.  I don't think the calculations are being performed incorrectly.  It is conceivable to me that the _MONOTONIC vs _REALTIME is potentially an issue, but I'll also note the following from stackexchange:

https://stackoverflow.com/questions/3523442/difference-between-clock-realtime-and-clock-monotonic
"Note that on Linux, CLOCK_MONOTONIC does not measure time spent in suspend, although by the POSIX definition it should. You can use the Linux-specific CLOCK_BOOTTIME for a monotonic clock that keeps running during suspend."

The sensor polling must be done in real time, if _MONOTONIC doesn't capture some portion of time then the results will not be correct.

mihu-ov, 2022-02-18:
Let´s get back to the beginning for a moment. The MS5611 pressure sensors used on the OV sensorboard (and many other paraglider, R/C model, etc. varios) are cheap, available and offer great resolution and low noise if used correctly. "Used correctly" means no light on the sensors, clean power supply, no RF energy injected on the hardware side, and precise timing of sensor reads on the software side. There is some speculation about the reasons why this precise timing is required and also some measurements on the web, most probably tracing the issue back to minor temperature differences in the sensor if the  time between two sensor reads is not constant.
When sensors are read by a dedicated micro controller there are usually no issues as it is easy to time sensor reads from a hardware clock and usually not much else is going on (tried this with my sensorboard, no issues at all). On a full blown Linux system getting the timing correct is hard to achieve. @linuxianer99 created a state machine in sensord to achieve proper timing. @hpmax did a lot of analysis on sensor noise etc. and tried to improve things. But at the end, it all comes back to how to achieve precise timing of sensor reads.

MaxKellermann, 2022-02-18:
@hpmax you're right, I mixed up `clock_nanosleep()` and `nanosleep()`, and got further confused by your code (ab)using the `curtime` variable for a relative time duration.

OK, so the practical problem is wrong sensor values because of MS5611 limitations. I understand.

On a multi-tasking operating system (unlike single-task microcontrollers), there is no guarantee your process ever gets scheduled. But you can improve your chances by using the real-time scheduler (`SCHED_FIFO` or `SCHED_RR`). Did you ever try that?

MaxKellermann, 2022-02-18:
> The sensor polling must be done in real time, if _MONOTONIC doesn't capture some portion of time then the results will not be correct.

Not relevant. The OpenVario never suspends. Even if it would, the difference would not matter.

hpmax, 2022-02-18:
@MaxKellerman There are two obvious answers in my book to solving this problem:

1) Use the MS5611 but have a dedicated chip (FPGA, or cheap 8-bit process, or possibly a 32-bit processor with a realtime OS, like ESP32) or....
2) Use the LPS22 sensor:  The LPS22 sensor can be configured to automatically strobe the conversion on a regular interval, and then to stick the results in a FIFO meaning it's a lot simpler to use.  The LPS22 is probably a bit noisier (although they have a new version out called the LPS22DF which might be quieter) but only a bit, but it is way easier to use and not worry about all the ugliness.  It's also considerably cheaper.  The MS5611 on Digikey is $8 in qty 100 (and it's closer to $9 on Mouser), versus $2.50 for the LPS22.  

So the new board features LPS22HH (maybe the LPS22DF) and an ESP32-S2 to run everything (I'll probably switch to the S3 in the next spin).  The goal is to make this self-contained so there won't be anything on OV, it'll just talk to XCSoar like any other vario -- although if it's embedded in the OV itself it may need to talk to OV via I2C (the new board has two I2C busses, an internal bus with all the sensors on it, and an external one which it talks to CubieBoard).

As for the realtime scheduler:  No, I didn't try that.  It is completely beyond my scope of knowledge.  I was unaware there was good realtime scheduling under Linux.

MaxKellermann, 2022-02-18:
> As for the realtime scheduler: No, I didn't try that. It is completely beyond my scope of knowledge. I was unaware there was good realtime scheduling under Linux.

All you need to do is add `CPUSchedulingPolicy=fifo` to the systemd unit.

If it's "good" or not depends on your point of view - nothing beats a dedicated CPU/microcontroller which will not ever be interrupted by a kernel. And dedicated real-time operating systems like QNX are probably more reliable than Linux's real-time scheduler. But I think we can tweak Linux so that it's good enough for OpenVario.

There's more we can (need to?) do than just enabling the real-time scheduler: currently, the kernel is compiled with a 100Hz timer interrupt; increasing that would probably help. And we should make the kernel preemptible.

sensord's timer slack is currently at the kernel's default value of 50us. That's good enough, isn't it? I see wait calls that are around 10ms.

mihu-ov, 2022-02-18:
There was / is? a similar issue on PX4, https://github.com/PX4/PX4-Autopilot/issues/7311 (with an interesting link to the temperature changes due to sensor read timing differences)
PX4 runs NuttX RTOS, so even with RTOS no final cure but good enough with an outlier detection.

@linuxianer99 built a small microcontroller (Cortex M0) board to interface between sensorboard and Cubieboard years ago but at the time we decided to go without. In my opinion, the state machine sensord was good enough for many people. If we can improve timing as @MaxKellermann suggested and with outlier detection the Kalman filter behind the raw sensor data should be able to do the rest. 

We need to cope with incorrect sensor data anyway as one issue we have no control of is RF injection in the sensors when transmitting on the radio. This depends on a lot of factors (OV housing, but mostly aircraft wiring, aircraft antenna cable, etc.) and can make the Kalman filter run away and not recover. I recently learned also on the new Hawk vario the filter can run away and needs to be reset due to bumps when closing the canopy, so at least we are in good company ;-)

MaxKellermann, 2022-02-18:
Running sensord for a while with `SCHED_OTHER` (the default scheduler at default priority):
```
voluntary_ctxt_switches:	24955
nonvoluntary_ctxt_switches:	709
```
Now with `SCHED_FIFO` (real-time):
```
voluntary_ctxt_switches:	66996
nonvoluntary_ctxt_switches:	6
```
With a real-time scheduler, there are only very very few non-voluntary context switches, as you see.
I tried keeping the CPU busy by zooming around XCSoar, but there was no noticable increase in the nonvoluntary value.
I think this one-liner solves the timing problem forever.

hpmax, 2022-02-18:
@mihu-ov I spent a lot of time trying to compensate for the error.  It's *REALLY* hard.  It's actually pretty easy to detect the problem because there's a simultaneous jump in both the pressure sensor and the temperature sensor and there is a tell-tale timing jitter which is readily detectable.  But this leaves a few issues:

1) It's not one sample that gets screwed up it could be a second or two's worth of samples.  Which means it's easy to detect when it starts, but not easy to detect when it ends.  And I've seen cases where it triggers a second time before it was finished the first event.
2) You could just blank the vario during the event, but again, you'd need to know when it ends, and I'm not sure this is real desirable behavior anyway.
3) It is theoretically possible to compensate, but it's REALLY hard.  You are basically adjusting the pressure reading by the difference in temperature reading on the other sensor... but it turns out that's a multidimensional function because it also depends on the pressure reading.

I'm not sure what the Kalman filter could do to filter it out.  First, whatever happens on one sense happens on both... that's what you'd expect in an actual thermal.  If only one adjusted then you could say, okay, I'm going to just use the static sensor right now instead of the TE, or I'm going to prioritize static and pitot sensors over TE, but that doesn't work.  The Kalman filter in this is actually pretty dumb.  It's ONLY looking at one piece of data, the TE pressure.  In the end it basically acts like a plain old regular filter.  You can adjust the Kalman gain, but that basically just acts like a time constant.  Maybe you could dynamically adjust Kalman gain based on detecting an event, but it gets ugly.

Has anyone documented that it's affected by RF and been unable to filter it by normal means?  My FSG-2T was actually hearing the USB traffic on the OV.  But I think the problem was the feedline on the FSG-2T was junk.  I'm in the process of replacing it with RG-400.

hpmax, 2022-02-18:
@MaxKellermann That's great, let's get it wrapped up into the release...  Like I said, I think whatever is affecting sensord also affects variod and pulseaudio because that's the only way I can explain the bufferunderruns.

closed 2022-02-18 by linuxianer99

mihu-ov, 2022-02-18:
@hpmax Yes, I understand this is difficult, probably impossible. We simply can´t re-create sensor data lost due to erratic measurements. What if we 

1) Reduce the chance of incorrect timing / measurements as already implemented by @MaxKellermann 
2) Just accept that the sensor data / vario is incorrect if there is no valid measurement (due to timing errors, due to RF) but
3) Reset the Kalman filter in case of a runaway so it goes back to normal operation when data is back

linuxianer99, 2022-02-18:
I have still the opinion, that a external µC is the best solution ...
If I have to choose now, I would you a ESP32 controller ... 2 CPU Cores - only can only do the sampling ...
Also a standalone sensorhub with WLAN is possible using such an architecture ...

MaxKellermann, 2022-02-18:
Before we discuss more complex hardware, can you first check if the problem is still present with real-time scheduling enabled?

hpmax, 2022-02-18:
@mihu-ov 

1) Yes, that's great!
2) I think there's not much choice.   However, as I said there's no good way to automatically suppress it because it goes back to normal slowly.    In the case of RF from a transceiver, at least the pilot is aware of it.  In the case of the timing issues,  there's no clear way to alert the pilot of bad data.  We could alter the protocol to provide an alert to XCSoar and XCSoar could display the alert for a short period of time.   But that's a change to XCSoar too.
3) Runaway only happens if my compensation scheme doesn't detect an end to it.  I really tried to make it drop out of compensation mode in a bunch of different ways so it'd be hard to get stuck like that. 

@MaxKellermann Ship kind of already sailed on the more complicated hardware.   At the urging of a bunch of people,  @DanD222 and I developed a new board.   I plunked down a few hundred dollars and I have four of them sitting on my bench.   I'm intending to develop it into a standalone instrument what it was designed to also function as a replacement for the current board.   At the request of @linuxianer99, it's got an ESP32 on it and should operate totally stand alone. 

This is actually really good for XCSoar, because people will be able to install the instrument I'm a club ship and users could link to it via BT on their phones/tablets.

I'm really excited that you were able to fix the timing problem, but I'm still planning to move forward with the new hardware because it has multiple advantages over the current version,  including full MEMS sensor suite, better integration,  the ability to operate independently if OV crashes,  etc.