Openvario/meta-openvario #279: openvario-image-testing doesn't build

bomilkar, 2022-02-18:
It can't find the requested version of imagemagick.

`
WARNING: imagemagick-native-7.0.10_25-r0 do_fetch: Failed to fetch URL git://github.com/ImageMagick/ImageMagick.git;branch=master;protocol=https, attempting MIRRORS if available
**ERROR: imagemagick-native-7.0.10_25-r0 do_fetch: Fetcher failure: Unable to find revision 8b4e00829eb84d4e7b4da11acf1f98f1e8166e5b in branch master even from upstream**
ERROR: imagemagick-native-7.0.10_25-r0 do_fetch: Fetcher failure for URL: 'git://github.com/ImageMagick/ImageMagick.git;branch=master;protocol=https'. Unable to fetch URL from any source.
ERROR: Logfile of failure stored in: /workdir/poky/build/tmp/work/x86_64-linux/imagemagick-native/7.0.10_25-r0/temp/log.do_fetch.689038
ERROR: Task (virtual:native:/workdir/poky/meta-openembedded/meta-oe/recipes-support/imagemagick/imagemagick_7.0.10.bb:do_fetch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1333 tasks of which 0 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  virtual:native:/workdir/poky/meta-openembedded/meta-oe/recipes-support/imagemagick/imagemagick_7.0.10.bb:do_fetch
Summary: There were 5 WARNING messages shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
`
Logfile of failure attached
[log.do_fetch.689038.zip](https://github.com/Openvario/meta-openvario/files/8095807/log.do_fetch.689038.zip)

MaxKellermann, 2022-02-18:
This needs to be fixed in [openembedded-core](https://github.com/openembedded/openembedded-core). Unfortunately, [it's very complicated to contribute patches to Yocto](https://www.openembedded.org/wiki/How_to_submit_a_patch_to_OpenEmbedded), tried that years ago, but I'm not in the mood to go through all the hoops anymore.

bomilkar, 2022-02-18:
My mistake.
Sorry!

closed 2022-02-18 by bomilkar