Openvario/meta-openvario #283: Sensorboard microcontroller board

iglesiasmarianod, 2022-02-20:
Guys, I'm working on an intermediate microcontroller board to get sensor data.
I'm using ESP32 and, so far, I'm getting data from sensors.

A new sensorboard is being developed with new sensord. I don't know if this is going to be backwards compatible. If it is not, it will force everyone to change sensorboard or get stuck with old sensord.
My idea is to plug a controller between sensorboard and mainboard, then add a config option to sensord to read data from the intermediate board.

hpmax, 2022-03-06:
@DanD222 and I designed a replacement sensorboard with an ESP32-S2 on it.  The new board is mechanically a drop-in replacement for the old one but has a lot of improvements including much newer/better sensors.  It'll require new software, but, so what...  If you're rolling a new custom board anyway, what's the benefit of this vs using the a new sensorboard?  I just don't like seeing the project split into so many versions.  In the end, that's not going to really help anyone.

linuxianer99, 2022-03-06:
I think both approaches are valid. 
- The board in between (i would call it sensor hub) has the benefit, that no new sensor board is ncessary (cost).
- The new sensors incuding the sensor hub is the best solution for building a new Openvario device.

Some toughts in my mind for both solutions:

- The update of the firmware of this sensor hub should be possible in the same way.
- The communication protocol used, should be the same.

@DanD222 , @hpmax , @iglesiasmarianod :
Maybe you can have a look to use the same software sources (the controller is the same, so this should be possible) ?

Should we open a new topic in the discussions ??

iglesiasmarianod, 2022-03-07:
That'll be nice @linuxianer99.
Sensorhub (as you named it :) ) is not meant to replace new sensorboard. It solves timing problems we have and provides some backwards compatibility. I will not run sensord or variod from this board and it will not be stand alone.
Points I believed might be important to discuss may include:

- variod, should be run in the cubieboard as deamon, run by xcsoar, or run in sensorboard.
- sensord. Same thoughts
- protocols How do you think it should work in the different configurations?
- XCSoar protocol if IMU data is included.
- updates. How do we update firmware through openvario

I can help with ESP32 code. For testing purposes I'm using Arduino IDE but. Final code will be written usinh to migrate to ESP-IDF and eclipse.

hpmax, 2022-05-30:
@iglesiasmarianod I have no experience with ESP32 coding... If you could give me a starting point, it would be much appreciated.

FWIW, my ultimate goal is to develop an external device that would talk to OV (or XCSoar) over Bluetooth and give you a fully independent device with a display.  I think this improves the reliability of the system versus relying on OV.

mihu-ov, 2022-05-31:
Maybe you want to look at https://github.com/iltis42/XCVario for a starting point?

hpmax, 2022-05-31:
@iglesiasmarianod I'm happy to work with you on an appropriate protocol.  I anticipate having:

GPS data (position, ground velocity), AHRS data (roll, pitch, magnetic heading), airspeed (indicated and true), static pressure, and compensated vario, outside air temp, humidity, and voltage.  

I could potentially provide uncompensated vario, and netto data as well.  I will try to compensate for magnetic deviation internally, but there is an additional question of should I compensate for magnetic variation.  Given GPS data and some knowledge of where the north pole is, I assume it shouldn't be hard, but given how fast the north pole is moving, I'm not sure there is an easy way to get that information from the Internet.  Best I could find is here:

https://www.ngdc.noaa.gov/geomag/GeomagneticPoles.shtml

I think most of this is actually covered in currently existing protocols.  We certainly have voltage, AHRS can be done through Levil protocol, GPS is GPS, temperature and I think humidity are in the current environmental protocol, etc.

Flap settings would be interesting, but I'm not sure how best to do a flap sensor.  My first guess, only because I like non-contact, and I think it would work on my LS6 is to attach a small reflective "target" via a hose clamp under my flap control rod and then mount a small optical time-of-flight detection sensor and an Arduino or some other microcontroller that would communicate via USB or BT to the OV directly.

lordfolken, 2022-05-31:
Please try to avoid inventing additional Protocols, XCVario/OpenVario does most of what you currently want. The only thing i couldn't find is voltage. 
https://github.com/XCSoar/XCSoar/blob/051adf385a61ece227360f91f70ea13096354b4d/src/Device/Driver/XCVario.cpp#L51-L71

hpmax, 2022-05-31:
@lordfolken As stated, I don't want to invent a new protocol if I don't have to.  XCVario protocol doesn't strike me as that comprehensive.

It lacks humidity,  voltage,  magnetic heading,  airspeed (true/ indicated) as well as roll/pitch/yaw rate.  That said,  most of these are covered by other protocols that XCSoar ALREADY covers.   The OV protocols have airspeed, voltage and I think humidity.   I believe Levil covers magnetic heading and rates.

iglesiasmarianod, 2022-05-31:
Hi guys.
My intention is not inventing a new protocol. I want to include IMU data into existing OpenVario $POV protocol so it can be used by sensord and variod.
I believe an improvement to sensorboard and, in the end, a new one with an MCU for new devices is the way to go.
This leads me to the question about what to do with sensord and variod:

- New sensorboard will calculate all data and produce sound if I'm not mistaken. This makes OpenVario daemons not necessary anymore and the new sensorboard is a variometer by itself. If it uses it's own protocol and this protocol gets included in XCSoar nothing needs to be done by OpenVario.
- Max proposed XCSoar to produce vario sound so we can ditch the audio mixer and Variod. We can use Sensord output to feed XCSoar. In that case we can still use OpenVario protocol and current sensorboards.
- I'm proposing the intermidiate solution. Keep deamons, solve MS5611 timing glitches and use IMU data from the already installed sensors adding an MCU. This makes this solution compatible with what Max proposes about Variod and all installed sensorboards out there.

hpmax, 2022-06-01:
@iglesiasmarianod To be clear, when I first discovered the flaw with the MS5611, my original proposal was to attach an AVR to the I2C port on the back of the sensorboard, and allow the AVR to poll all the sensors and shove the data into a FIFO, and then have sensord read out the FIFO through the i2c.  My concern with this is sensord would have to know when to talk (and when not to talk) to the AVR.  Without a separate way to communicate to the AVR (or in your case the ESP32) it's probably a bit tricky.

The new sensorboard does have a Class AB audio amplifier on it.  It is setup to allow creation of an audio variometer mixing of audio from the Cubieboard either with analog circuitry, or potentially digitally via the ESP32 with the vario audio.  Honestly, I think it's a crappy solution.  If Max can integrate the audio into XCSoar and have it work well, I'd prefer that, but... one of the points behind making a separate device is that it could operate properly even if XCSoar/OV crashed or otherwise had problems.