Openvario/meta-openvario #287: build fails to fetch sources

bomilkar, 2022-04-01:
Following the instructions in README.md running

`git clone --recurse-submodules https://github.com/Openvario/meta-openvario.git`

results in the following error:
......
Cloning into '/media/disk2/OV_build_2022_master_7.24rc1/meta-openvario/meta-sunxi'...
fatal: remote error:
  **The unauthenticated git protocol on port 9418 is no longer supported.**
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
fatal: clone of 'git://github.com/linux-sunxi/meta-sunxi.git' into submodule path '/media/disk2/OV_build_2022_master_7.24rc1/meta-openvario/meta-sunxi' failed
Failed to clone 'meta-sunxi'. Retry scheduled
....

The error message "The unauthenticated git protocol on port 9418 is no longer supported" seem to indicate the issue.
But I don't know where and how to fix this.

lordfolken, 2022-04-01:
The url in the submodule needs to be changed from git to https or git+ssh.

bomilkar, 2022-04-02:
I can manually install the offending module:
`git clone https://github.com/linux-sunxi/meta-sunxi.git`
That goes through, but it doesn't seem to be enough:  there is no `openembedded-core/oe-init-build-env
`
I don't understand when the directory `openembedded-core` is supposed to be populated. Mine is empty at that stage.

Do I have to "repo init" after fetching the repos?

lordfolken, 2022-04-02:
https://github.com/Openvario/meta-openvario/blob/master/.gitmodules 
list all the submodules.The submodules are other projects and are not part of the meta-openvario projects. 

You can manually run submodule init && submodule update after correcting the url to github.

bomilkar, 2022-04-02:
Thanks, this does the trick!
Assuming PR #288 gets merged I close this.

closed 2022-04-02 by bomilkar

linuxianer99, 2022-04-02:
@lordfolken  Thanks for the quick PR !

PR #288 (2022-04-02): use https for github based submodules
fixes #287