Openvario/meta-openvario #289: chattr comand not found

Blaubart, 2022-04-02:
chattr is no longer implemented at OV image. It is pretty usefull for club gliders.

linuxianer99, 2022-04-05:
Can make a PR please ??

MaxKellermann, 2022-04-05:
Why is chattr useful? I don't think the OV base image needs to include each and every exotic tool.

Blaubart, 2022-04-05:
I agree with you Max, but we use this tool to lock the XCSoar profile on our club planes. We all know it, everyone fiddles with the device and in the end nothing works anymore. And then we have work to get everything up and running again over the weekend. Withe chattr XCSoar always runs in the fixed version. If certain members want to use different settings, they get their own profile. This makes using the OpenVario in clubs much easier!

Blaubart, 2022-04-05:
by the way, i2cdetect has also disappeared from the image. The program is mandatory if you build your OV by your own and want to check whether you have soldered all the sensors correctly.

linuxianer99, 2022-04-06:
@Blaubart What are you doing in detail with chattr ?
Can is be done with chmod ??

Blaubart, 2022-04-06:
chmod does not, as far as I can tell, make a file immutable. chattr can do that. With this you are able to make the XCSoar profile in club aircraft unchangeable, so that the OpenVario always starts in the same configuration.
Otherwise the system will be fiddled with unnecessarily, nothing will run reliably and acceptance in the club will be low. Not to mention the work that people like me have to do to fix everything!

linuxianer99, 2022-04-06:
what about chmod and remove the "write" attribute ?
Sorry, but I don't see why chattr is needed ...

Blaubart, 2022-04-06:
Hmmm, I will give a try!

Blaubart, 2022-04-06:
I tried chmod 555 openvario.prf and it dosen't work. I was able to change the profile and after reboot, changes are still there!

closed 2022-05-31 by Blaubart