Openvario/meta-openvario #291: Compiling sensord code

hpmax, 2022-04-06:
I just rebooted my development machine after a long time, and found that the old instructions to start the build flow don't work.  I used the new instructions and it seemed to work fine, however, when I built sensord and uploaded it to the OV, I got: "sensord: /lib/libc.so.6: version 'GLIBC_2.34' not found (required by sensord)"  I figured maybe the new code base requires a newer version of the OS, so I tried to use the latest 21118 release image and then the 22005 development snapshot, everything reported the same error.  

Anyone got a solution for this?

linuxianer99, 2022-04-06:
Where do you get the 22005 development snapshot ??
Openvario never released such a snapshot ..

Is the error also happing, with a complete image build from the lastest sources ?

hpmax, 2022-04-06:
I got the development snapshot from: https://openvario.org/doku.php?id=downloads

I have not attempted to build the complete image I assume I just run bitbake on the whole thing rather than doing the -c devshell?

At this point, I really want to be able to just have something that allows me to compile code as I could before.  If I recall, using the old container caused it to point to files that either weren't there or were no longer accessible which is why I switched to the new one.

linuxianer99, 2022-04-07:
So from 5th of January up to now, a lot of improvements have been done in the image.

The only way is to update the whole image ...

hpmax, 2022-04-07:
I don't understand what you mean, I re-flashed the 22005 image and then uploaded my compiled code to it.  Are you telling me that the 22005 image doesn't match what make is putting out?  If so, is there an image file I can flash that will work?  We really should have a published image that matches the current development environment.

linuxianer99, 2022-04-07:
I am telling you, that lots of improvements have been done since 22005 and there was no release up to know.
I also don't know, who did the 22005 snapshot ... 

so either use the warrior based image (which is old) or compile a new image on your own, if you want to have the current bleeding edge ...

hpmax, 2022-04-07:
So there is no current working image available for download?

How do I use the warrior image?  When I tried to run docker to start the devshell it failed.

linuxianer99, 2022-04-07:
The lastest release is 21118.

Go to the warrior branch and use the guideline in this branch.

hpmax, 2022-04-07:
I am following the directions here: https://github.com/Openvario/meta-openvario/blob/warrior/README.md
However, when trying to use repo, I get the following:
"The unauthenticated git protocol on port 9418 is no longer supported." 

I tried altering: repo init -u git://github.com/Openvario/ovlinux-manifest.git -b warrior
to: repo init -u https://github.com/Openvario/ovlinux-manifest.git -b warrior

But that didn't help too much

linuxianer99, 2022-04-07:
Have a look here #288

hpmax, 2022-04-07:
I am thoroughly confused... It appears that according to #288 the branch has been updated to use https:// instead of git:// but even when I do a repo init -u https://github.com/Openvario/ovlinux-manifest.git -b warrior it still appears quite broken.  Is that the wrong file to be doing the init on?

linuxianer99, 2022-04-07:
I did a quick backport in ovlinux-mainfest.
Just pull again .. and give it a try ...

hpmax, 2022-04-07:
repo init appeared to work.
repo sync did not:
"Fetching projects: 25% (1/4) pokyfatal: remote error:
The unauthenticated git protocol...."

linuxianer99, 2022-04-07:
Which repo ??
Can you please post the complete log ...

hpmax, 2022-04-07:
As previously stated, i am following the directions, here: https://github.com/Openvario/meta-openvario/blob/warrior/README.md

dan@dan-OMEN-by-HP-Laptop:/home/clean2$ mkdir workdir
dan@dan-OMEN-by-HP-Laptop:/home/clean2$ repo init -u https://github.com/Openvario/ovlinux-manifest.git -b warrior

Your identity is: Dan <me@me.org>
If you want to change this, please re-run 'repo init' with --config-name

repo has been initialized in /home/clean2
dan@dan-OMEN-by-HP-Laptop:/home/clean2$ repo sync
Fetching projects:  25% (1/4) pokyfatal: remote error: 
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
fatal: remote error: 
  The unauthenticated git protocol on port 9418 is no longer supported.
Please see https://github.blog/2021-09-01-improving-git-protocol-security-github/ for more information.
error: Cannot fetch meta-openvario from git://github.com/Openvario/meta-openvario
Fetching projects:  75% (3/4) meta-openembeddedremote: Enumerating objects: 3588, done.        
remote: Counting objects: 100% (375/375), done.        
remote: Compressing objects: 100% (272/272), done.        
remote: Total 3588 (delta 174), reused 231 (delta 85), pack-reused 3213        
Receiving objects: 100% (3588/3588), 925.23 KiB | 4.56 MiB/s, done.
Resolving deltas: 100% (1626/1626), done.
From https://github.com/linux-sunxi/meta-sunxi
 * [new branch]      bananapi-fix     -> sunxi/bananapi-fix
 * [new branch]      daisy            -> sunxi/daisy
 * [new branch]      danny            -> sunxi/danny
 * [new branch]      dizzy            -> sunxi/dizzy
 * [new branch]      dora             -> sunxi/dora
 * [new branch]      dunfell          -> sunxi/dunfell
 * [new branch]      dylan            -> sunxi/dylan
 * [new branch]      fido             -> sunxi/fido
 * [new branch]      hardknott        -> sunxi/hardknott
 * [new branch]      honister         -> sunxi/honister
 * [new branch]      jethro           -> sunxi/jethro
 * [new branch]      krogoth          -> sunxi/krogoth
 * [new branch]      master           -> sunxi/master
 * [new branch]      mbe-bananapi-m64 -> sunxi/mbe-bananapi-m64
 * [new branch]      morty            -> sunxi/morty
 * [new branch]      pyro             -> sunxi/pyro
 * [new branch]      rocko            -> sunxi/rocko
 * [new branch]      sumo             -> sunxi/sumo
 * [new branch]      test-atf         -> sunxi/test-atf
 * [new branch]      thud             -> sunxi/thud
 * [new branch]      warrior          -> sunxi/warrior
 * [new branch]      zeus             -> sunxi/zeus
Fetching projects: 100% (4/4), done.
Checking out projects:  50% (2/4) meta-openembeddederror: Cannot checkout meta-openvario: ManifestInvalidRevisionError: revision refs/heads/warrior in meta-openvario not found
error: in `sync`: revision refs/heads/warrior in meta-openvario not found
dan@dan-OMEN-by-HP-Laptop:/home/clean2$

linuxianer99, 2022-04-07:
I changed default.xml again ...
Can you give it another try ?

hpmax, 2022-04-07:
That appears to have worked.  I also appear to have successfully built a 22097 image.  I really think we should be publishing development build images as new versions come out, especially if the released images cannot work with software compiled under the current build platform.

lordfolken, 2022-04-07:
We should change those urls also. 
Also in the other branches such as warrior we will have the same issue.

linuxianer99, 2022-04-07:
@hpmax 
Feel free to take this task of always publishing these images ;-)

linuxianer99, 2022-04-07:
So i do this on my development machine and upload in the github here.
But the better way will be, to use the github actions for automation.
I am working on this since a few weeks, but it has some "special" problems with our big builds ...

Release notes are generated using gren https://github.com/github-tools/github-release-notes

linuxianer99, 2022-05-30:
@hpmax Is this issue still open ?

closed 2022-05-30 by hpmax