Openvario/meta-openvario #294: Installation of a damping for the Vario

Blaubart, 2022-04-15:
I'm not sure whether my suggestion should be implemented in OpenVario or in XCSoar. But the vario that can be displayed next to the NavBoxes is very restless. Some seem to be bothered by that. Would it be an option to add adjustable damping?

hpmax, 2022-04-28:
The config file for sensord should contain a Kalman filter gain, by adjusting this you should be able to make it respond slower.  There should be a better approach, but that's there now.

Blaubart, 2022-04-28:
Thanks! That would at least be a temporary solution. Unfortunately, since I have absolutely no idea about it, I need a more detailed description of how I could try it out.
Thank you very much!!

Blaubart, 2022-04-28:
Thanks, I will try

hpmax, 2022-04-28:
I can't recall for sure,  but I think it should be: vario_config>0.  I don't think there is any other restriction.

mihu-ov, 2022-04-28:
You probably need to setup some kind of simulation to find meaningful values. Trial and error in flight doesn't work well.

hpmax, 2022-04-28:
I think you can record data and replay it.  There's no mechanism to adjust vario_config in flight.   You'd have to edit the file and restart sensord.

closed 2022-05-31 by Blaubart