Openvario/meta-openvario #295: Is i2c communications working?

hpmax, 2022-04-28:
I have been trying to talk to the LSM6DSM on the next gen sensor board with sensord, and it is experiencing erratic behavior.  The behavior involves digital data from the FIFO getting out of order or screwed up.  I have been working on debugging this for a good long while to no avail.  I have discovered that the problem is NOT specific to the board.  I replaced the sensorboard with a generic dev board that only had an LSM6DSL, it still had the same issue.  I have a scope on the SDA and SCL pins and can confirm the decoder in the scope says that I'm writing what I think I'm writing and I'm reading what I think I'm reading...  I also know that the problem occurs after repeated accesses to the FIFO, perhaps 1000, perhaps 10k, perhaps 100k, perhaps 200k, but not more than that -- but it is not time related.  I have been in contact with ST and they executed similar code on their boards and claimed that it doesn't have the problem I am seeing.  I can issue a software reset and it will fix itself for a while before it breaks again.

They are claiming it is a software issue.  Another person I talked to, Kris Winer, said the LSM6DSM works great and suspected that my i2c drivers may be questionable.  I am including some scope traces.  Blue is SCL, Yellow is SDA.  There is decoding at the bottom.  What you'll see is I read out address 0x3c and get 0x2 (which indicates the next FIFO read should be GyroZ).  I then read out the FIFO and get 0x32 0x00 (0x32 is LSB, 0x00 is MSB) this should be about 0.2 degrees per second -- not great but at least it's plausible.  I then read out address 0x3c again.  It should be 0x03, but it comes back as 0x01. 

I'm wondering if there is any chance that this is an issue with however Linux talks to the i2c bus. 
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

hpmax, 2022-04-28:
I spoke to an ST App Engineer and he confirmed the i2c bus communication is screwed up.  Here's what he said:

"Each I2C transaction should start with a I2C start condition. which is correct in your case
On multiple byte read, each byte read should be followed by an ACK from the master. when you don't want to read more data, master should generate NAK signal on the last byte read indicate the end of read. And master should not send out more invalid clock signals. A STOP should be issued from the master that the communication is over.

In your read operation, you didn't give NAK to slave, and keep sending out the clock without the STOP, so the sensor will continue output the next FIFO sample. Even though the clock is not complete, this is why there is a question mark from your oscilloscope decoder. The signal doesn't comply with I2C protocol and cannot be decoded, and when you try to read the next sample, you have jumped over one sample or one byte, so the data reading is misaligned."

Any comments on what we're doing wrong here and how to fix it since I'm pretty sure the i2c communication is pretty much the same throughout the whole project.

hpmax, 2022-04-30:
I changed the code to use: ioctl(sensor->fd,I2C_RDWR,&operation) 
rather than ioctl(sensor->fd,I2C_SLAVE,i2c_address)

and it seemed to work...  I'm currently on 3.7 million reads without any corruption whereas I was usually seeing corruption before 150k prior to the change.  This makes me wonder how much of the i2c access software is goofy and the sensors are just insensitive to the poorly formed communication.

hpmax, 2022-04-30:
I believe the issue is that this particular device wants to see a write immediately followed by a read, which must be done with I2C_RDWR.  It should be a bit faster, but it's unclear if it's actually necessary for the other devices.

linuxianer99, 2022-05-30:
@hpmax Is this issue still relevant ?