Openvario/meta-openvario #305: XCSoar profile lost after abrupt power off #90  -- still existing?

xmeissn, 2022-05-11:
Release 22086 ; File: OpenVario-linux-openvario-image-glibc-ipk-22086-openvario-7-CH070.rootfs.img;
I have just yesterday noticed exactly that in the closed issue #90 described behaviour while turning off OV by abrupt power off.  We have two profiles on our device....each one was turned to zero after abrupt power off. After starting OV again everything was white....configuration completely lost. Is this issue really fixed?

lordfolken, 2022-05-12:
So the solution for #90 will not solve the problem here. As for #90 have an effect, xcsoar needs to exit. 

So settings should be flushed to disk since several years:
https://github.com/XCSoar/XCSoar/commit/e6382171be2baea2fdb5fba746b045aa37ebd675

We also don't seem to have an openfile handle on the profile file. At least lsof doesn't show one. So this seems to be handled correctly. 

What confuses me is that "both" profiles are 0 bytes. So even the one that wasn't active?

lordfolken, 2022-05-12:
We probably should have a look at the sysctl settings for vm.dirty*
Here is a good explanation:
https://lonesysadmin.net/2013/12/22/better-linux-disk-caching-performance-vm-dirty_ratio/

Also I still vote to put the home folder in another partition from the system and potentially mount with sync.

MaxKellermann, 2022-05-12:
Don't mess with VM settings. The problem here is that XCSoar truncates the file instead of creating a new one and linking it atomically. To be safe against power failure, we can use fsync().

MaxKellermann, 2022-05-12:
This is how to use fsync: https://github.com/XCSoar/XCSoar/commit/a9e7334c23ebbaa50f511931fcddea493adf92cb

lordfolken, 2022-05-12:
Thank you! This might also solve https://github.com/XCSoar/XCSoar/issues/838

xmeissn, 2022-05-13:
> 

> So the solution for #90 will not solve the problem here. As for #90 have an effect, xcsoar needs to exit.
> 
> So settings should be flushed to disk since several years: [XCSoar/XCSoar@e638217](https://github.com/XCSoar/XCSoar/commit/e6382171be2baea2fdb5fba746b045aa37ebd675)
> 
> We also don't seem to have an openfile handle on the profile file. At least lsof doesn't show one. So this seems to be handled correctly.
> 
> What confuses me is that "both" profiles are 0 bytes. So even the one that wasn't active?

Sorry, that might be reportet from me in an misunderstandable sense....only the in this moment ACTIVE profile was made empty. After that, i started new and changed to the other one with the same procedure.

linuxianer99, 2022-05-30:
As far as i understand, this is no Openvario problem, but XCSoar issue ??

lordfolken, 2022-05-30:
Yes this is an XCSoar issue. This should no longer happen in a build based on current XCSoar master.

closed 2022-05-30 by linuxianer99