Openvario/meta-openvario #309: Im willing to help out as well testing kernels once I get back from work. Should be after the weekend. I do need a guide though how to build it on Windows 10.

tb59427, 2022-05-25:
Im willing to help out as well testing kernels once I get back from work. Should be after the weekend. I do need a guide though how to build it on Windows 10.

_Originally posted by @hph304 in https://github.com/Openvario/meta-openvario/issues/304#issuecomment-1137677475_

I'm afraid I am of no help when it comes to windows. Have been using Macs since 1984 and Unix boxes starting with Version 7 Unix in 1986. Have never used Windows for longer than 10mins in a row. I guess you have to get a VM and install the latest Ubuntu (e.g.) on it and then follow the instructions here on git.

tb59427, 2022-05-25:
Hey admin, please delete. Mistake on my side when trying to quote...

closed 2022-05-25 by tb59427

linuxianer99, 2022-05-30:
@tb59427 Delete the whole issue ?

tb59427, 2022-05-30:
Yep. I created it wrongly when I only wanted to reply to hph304