Openvario/meta-openvario #312: OV control stick freezes in flight

tatro77, 2022-06-07:
Hi

I have been testing my new Stefly OV 7-CH070 with out sensor board for several weeks. Unfortunately I get a recurrent problem where the control stick stop working after approximately 1:30 hours in flight(12,7 battery voltage). 
If I reboot the system in flight the stick control starts working again. 
I have installed the rotary module and noticed that the OV is not frozen when the failure occurres. 
After the first time the problem occurred, I did manage to fix it after landing by unplugging and plugging again the usb connector of the control stick with the OV stil on.

I hope you can advice me how to resolve this problem. 
I have tried several options: 
I installed a new Stefly control stick circuit board, cables and usb connector (to rule out the possibility of a defective part).  
I tried the OV image 22086 and the official (currently using 21118,XCSoar-7.4)
I tried to flash 3 different brands of micro sd cards 
I connected the control stick directly to the OV USB port without any USB hub.

I have run out of ideas and could use some expert advice from you.

The full system configuration is
 
- single seat glider without engine
- 7" Stefly OV no sensor board.
- Stefly Rotary Module and control stick
- Lx nav S10 connected to ttys1 
- Classic alarm F9 connected to ttys3 

Thank you in advance.

franke1312, 2023-01-12:
Hi Tatro,

i had exactly the same issue in our LS8. I tried to a lot of things, but it didnt help. Then one of our club member told me to change the delivered usb cable of the stefly stick with shielded USB cable. I bought a USB cable and changed this cable ith the delivered stefly calbe. Now it works fine!

Regards
Alex

tatro77, 2023-01-13:
Hi AlexThank you so much for the advice! I have tried all kinds of things but no luck. I installed the OV last Image hoping the problem to go away but had my doubts.I will install a shielded usb cable and try again. Best regards JuanSent from my iPhone

Scumi, 2023-04-06:
Well...after years of flying with the Openvario and the SteFly Stick, today the Stick remote control froze several times. Luckily I also have the rotary encoder installed, which still worked, so I could continue flying. Very weird, will do some troubleshooting soon. I have installed a new radio and a canopy strobe, so maybe it has something to do with that.

tatro77, 2023-04-11:
Hello. It is interesting to mention that I installed Stefly OV, the new AIR COM and Air control Display from Air Avionics at the same time. I managed to replicate the problem last season, pressing the PTT button multiple times. It has not been easy to consistently replicate the flaw. I have now switched to shielded cables for the PTT and so far it works fine. I haven't flown enough to be sure it's a complete solution, but it might be a good idea to try that or Alex's advice to use a shielded USB cable.
Regads 
Juan

Scumi, 2023-04-11:
Interesting, I use the same setup since this season. Unfortunately I cannot easily install shielded cables, at least not for the USB cable. But at least this gives me a hint what to look for. I will probably start with using shielded cables for PTT and hope it goes away.

Scumi, 2023-05-02:
Did some minor improvements, but they did not solve the problem.

Additional observations:
- Rotary control still works after the stick stops working
- A bluetooth keyboard also does not connect anymore

I also got a log from the event where the stick stops working:

```
Apr 06 10:27:28 openvario-7-CH070 kernel: usb 1-1.1: reset full-speed USB device number 3 using ehci-platform
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: can't restore configuration #1 (error=-32)
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: USB disconnect, device number 3
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.3: reset low-speed USB device number 5 using ehci-platform
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: new full-speed USB device number 7 using ehci-platform
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: New USB device found, idVendor=2341, idProduct=8036, bcdDevice= 1.00
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: Product: Arduino Leonardo
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: Manufacturer: Arduino LLC
Apr 06 10:27:29 openvario-7-CH070 kernel: usb 1-1.1: SerialNumber: HIDFG
Apr 06 10:27:29 openvario-7-CH070 kernel: cdc_acm 1-1.1:1.0: ttyACM1: USB ACM device
Apr 06 10:27:29 openvario-7-CH070 kernel: input: Arduino LLC Arduino Leonardo Keyboard as /devices/platform/soc/1c14000.usb/usb1/1-1/1-1.1/1-1.1:1.2/0003:2341:8036.0005/input/input11
Apr 06 10:27:30 openvario-7-CH070 kernel: input: Arduino LLC Arduino Leonardo Mouse as /devices/platform/soc/1c14000.usb/usb1/1-1/1-1.1/1-1.1:1.2/0003:2341:8036.0005/input/input12
Apr 06 10:27:30 openvario-7-CH070 kernel: hid-generic 0003:2341:8036.0005: input: USB HID v1.01 Keyboard [Arduino LLC Arduino Leonardo] on usb-1c14000.usb-1.1/input2
Apr 06 10:27:30 openvario-7-CH070 systemd-logind[198]: Watching system buttons on /dev/input/event4 (Arduino LLC Arduino Leonardo Keyboard)
Apr 06 10:28:45 openvario-7-CH070 ovmenu-ng.sh[259]: libinput error: event2  - Arduino LLC Arduino Leonardo Keyboard: client bug: event processing lagging behind by 13ms, your system is too slow
Apr 06 10:29:13 openvario-7-CH070 ovmenu-ng.sh[259]: libinput error: event2  - Arduino LLC Arduino Leonardo Keyboard: client bug: event processing lagging behind by 29ms, your system is too slow
Apr 06 10:29:13 openvario-7-CH070 ovmenu-ng.sh[259]: libinput error: event2  - Arduino LLC Arduino Leonardo Keyboard: WARNING: log rate limit exceeded (5 msgs per 60min). Discarding future messages.
```