Openvario/meta-openvario #314: OV is making a lot of RF noise

hpmax, 2022-06-18:
Yes, this is a hardware issue, not software, not sure where to post about hardware issues other than here.  My Stefly OV is producing a lot of RF noise that messes with my radio (Dittel FS2T).  Most of the noise is pretty narrowband, but splattered all over the place, 119-120.5 MHz seems particularly bad.  On 120 MHz, I recorded the following...  This seemed pretty unique, most noise sounded more white.

https://user-images.githubusercontent.com/3003859/174419270-d71ef8f1-1207-4ffa-a6bc-d5b20ce85fe3.mp4

The noise is worse the more that is plugged into the USB, but I can have noise from just the computer itself.  Here's what I have connected:

My OV is connected to: the rotary encoder device, and a 4 port hub.  The 4 port hub has: WiFi, GPS, a second hub under my seat, and the stick remote.

The second hub is a powered hub.  It has a TTGO T-Beam (running SoftRF), a BT dongle and a CP2101/CP2101 to talk to a Stratux.

I've done the following to try to mitigate noise:

1) Replaced the RG-58 in my cockpit with RG-400.
2) Installed USB isolators
3) Replaced the 12V->5V converters (for powering the powered hub, Stratux and my camera) with custom made isolated supplies with multiple layers of EMI filters.
4) Installed clamp-on ferrite beads (with multiple passes through the core and I made sure that they gave a LOT of isolation in the passband of the radio.  Ferrite beads were placed on all power supplies and all USB cables.

It should be noted that the Dittel is a single conversion superheterodyne receiver with a 21.4MHz IF, but since the noise is narrowband and varies with frequency, it seems much more likely to be in the passband than the IF.
 
I'm literally out of ideas.  The only thing I can think to do is replace the power supply in the OV with a better filtered or isolated supply, or add more capacitors on the IO to ground.  Say 250-1000pF to ground on anything that isn't supposed to have high speed signaling on it?

kedder, 2022-06-18:
Did you try separate battery for ov and radio?

hpmax, 2022-06-18:
@kedder I didn't try that, but I suppose it wouldn't be that difficult to pull the wires off and try it.  Implementing it would be a complete pain in the butt though.

DanD222, 2022-06-18:
I’d probably try kedder’s suggestion first, then changing the PSU in the OV.. then I’d probably redo the wiring to two-conductor (12V, GND) plus shield, with the shield being connected to a central ground point at one end only. A a major pain in the butt :(

1000pF to ground on every interface pin got rid of radio interference on my Borgelt vario, that might be worth a try too.

hpmax, 2022-06-18:
Definitely first step is kedder's suggestion.  Although if it works, I'm not sure what to do after that.  Time for the 1000pF caps?

Using shielded cable would indeed be a major pain.  Not sure what the Stefan is using for a power supply -- or if he's using 12V for anything or if it's just getting converted to 5V and being used there.

As for 1000pF... I think we have a mutual history with the 1000pF caps!  Did you mean 1000pF on the radio or the OV?  Either way, that's one of those things the factory probably should have done.  I guess I can ask Stefan for schematics.  I actually think 1000pF might be a bit too high.  When I looked at Kemet sim on a random 0603 50V 1000pF cap it showed the resonant frequency was at about 150MHz -- probably okay, but seems close to marginal.

Other than USB (which I don't think you could put anything on), there's power, two audio cables going into the radio, and one to the OV.  There's a PTT going to the radio, and that's it.  Other than the temp sensor and the I2C interface (I'm using the former but not the latter).

mihu-ov, 2022-06-18:
Try without anything that has an Arduino inside (stick remote, rotary encoders). 120MHz is a multiple of the 8MHz Arduino clock.

hpmax, 2022-06-18:
I am getting some noise even without anything plugged in.  However, that sound recording may have been made with one of the Arduinos plugged in specifically.  If that's the case it will presumably wipe out 120 and 128MHz and that's it.  Probably not the end of the world.  But it does beg the question if adequate RF-suppression was incorporated in those components.

A lot of the noise seems pretty intermittent, but that was pretty crystal clear.

kedder, 2022-06-18:
TBH I have no clue if my suggestion helps or no. But since you can measure the noise, I'd be interested to know :)

hpmax, 2022-06-19:
@kedder I tried your suggestion...  Ran the radio off my bench switching supply everything else off the battery.  There was no change.

I also added clamp-on ferrites over all the audio cables, and I can't tell that that made a difference.

One odd thing I noted...  The squeal I recorded wasn't present on 120MHz this time.  I then plugged in the cable to my SoftRF/TTGO T-Beam and the noise started...  But here's the kicker.  Then I unplugged it, and it didn't stop.  Now I'm totally at a loss.

linuxianer99, 2022-06-23:
Just a guess, but i think that is caused by USB. If nothing is connected, the USB controller is no active .. after the first connection, it stays active, even after disconnect ...
May you try another USB device instead the SoftRF ? Maybe a keyboard ...

lordfolken, 2022-08-23:
I also noticed the usb/rf problems. As soon as i plug-in a keyboard the rf noises spike.

hpmax, 2022-08-23:
There are probably a lot of sources of noise, but USB is definitely the worst one.  I think a lot of the noise is coming from the OV itself and is being transmitted through the USB cables.  Obviously, having the USB cables go through a separate (metal) conduit would be helpful.  I have tried using ferrite chokes, and found that they do reduce the problem but not eliminate it.

I don't know if the problem is related to the CubieBoard and if a RPi would be better (or worse).  I almost wish we could just communicate via Bluetooth, it seems safer from an RF perspective.

I think Timo might be on to something about the USB controller, but I don't know what you do about it (other than switch to a RPi or a different board).

This is just disappointing.  I feel like if I knew what I know now when I installed the OV, I would have done it very differently.

lordfolken, 2022-08-23:
i"m also not so sure about the rj45 jacks. When i installed the official stefly cable between my ov and an lxnavs80 i had a lot of interference. 

> I don't know if the problem is related to the CubieBoard and if a RPi would be better (or worse). I almost wish we could just 
> communicate via Bluetooth, it seems safer from an RF perspective.

Hah. After spending a day on my glider trying to eliminate the rf noise, this was my thought exactly to make things communicate via bluetooth instead. 

I in the end threw out the lxnav s80 and put in the XCVario.. that made it better.

But in general, i would love to move to a RPi. Personally i don't need a sensor board. And the screen would be better connected via HDMI.

But hten again i just found this: https://forums.raspberrypi.com/viewtopic.php?f=28&t=273373 and at 100MHZ too.

MaxKellermann, 2022-08-23:
"7 Colour E-Ink Display For Raspberry Pi": https://core-electronics.com.au/guides/colour-e-ink-display-raspberry-pi/

hpmax, 2022-08-23:
@MaxKellermann That's very cool, but they seem incredibly slow at updating, I'm not sure how well XCSoar would run on it (also would be nice to have a 7" model).  I do like that it's controlled over SPI which would make it easy to attach to either a Pi or a CubieBoard.