Openvario/meta-openvario #323: bitbake openvario-image-testing fails because sox is missing

bomilkar, 2022-07-28:
Recent versions of xcsoar require sox. Here is a snippet of the error message when `bitbake openvario-image-testing`

```
|   SOX     output/data/sound/beep_bweep.raw
| make: sox: No such file or directory
| make: *** [build/resource.mk:156: output/data/sound/beep_bweep.raw] Error 127
```
Please note `bitbake openvario-image` doesn't need sox and therefore is not affected.

closed 2022-07-28 by linuxianer99

Scumi, 2022-07-28:
Should we then add "--pull=always" to the docker command in the description?

linuxianer99, 2022-07-28:
Think this generates unnecessary load ... 

In my mind, i would have a look in the issues, if somebody has reported the problem already and if there is a resolution, before creating a new issue ...

bomilkar, 2022-07-28:
I'm sorry when I haven't seen that the issue had been reported already. Out of curiosity: how could I have found it??

Here is what I did when bitbake failed:
I built xcsoar to realize sox was missing. The developers manual mentions sox under the Requirements. Adding the package is straight forward in the native linux environment.
However, the docker system seems quite convoluted to me. I noticed "which sox" didn't find anything and poking around where to add it didn't lead nowhere.

@Scumi : thanks for the **constructive** hint! Adding "--pull=always" does the trick (and it didn't add significant load).