Openvario/meta-openvario #332: Multiple Image Build Issues

hpmax, 2022-09-24:
I tried to do some software development and found that it was complaining that GLIBC_2.34 couldn't be found, so I went to go look for a new image, but the latest one was middle of last year.  So I went to build a new image.  Found a few issues:

1) It complained that it wasn't able to read colors.xml (this was a warning)
2) It complained it couldn't read a number of files in xcsoar7.26-r0/git/output/data/sound including: beep_bweep.raw, beep_clear.raw, beep_drip.raw, fail.raw, insert.raw, and remove.raw.  This was an error and halted the bitbake.  I created bogus files and it continued the build and completed successfully.

I flashed a new image and booted off it.  I believe I successfully configured my wifi using connmanctl as I had previously, but when I went to ssh into it.  I got: "connection refused." I was able to ping it though.

I tried ssh'ing into localhost from the terminal.  That also gave me connection refused.  I have run ps | grep ssh, as well as netstat, and I can't find any evidence sshd is being run or even on the system.  There is no /etc/ssh directory and /usr/sbin/sshd isn't present.

Can someone please fix this or tell me what's going on.  It's kind of a showstopper for me.

realtimepeople, 2022-09-26:
Hi,
Somehow the tool sox is missing.
Does anyone know how to add it ?

eku60, 2022-11-10:
The problem is still actual. Tried to build an image on November, 6th receiving the same (unable to read colors.xml).
Besides meta-openvario, I cloned XCSoar (7.29, the trunk) and was able to build and run XCSoar on ubuntu und run it.

I used a new clone of meta openvario from git.

Regards eku60