Openvario/meta-openvario #335: Remove blank values in Profile

lordfolken, 2023-02-08:
https://github.com/Openvario/meta-openvario/blob/5108189283e2ebccdf602738775832e2fbb01be0/meta-ov/recipes-apps/xcsoar/xcsoar-profiles/openvario.prf#L3

I think all the blank settings ("") should be removed. That way if they are externally defined, they are not overwritten. 
We also should be very clear which settings are needed from an OpenVario perspective.

mihu-ov, 2023-02-08:
Good point. 

The idea with including this profile was to help a first time user with testing the system after the first boot up. Include an opernvario.xci file with definitions for a remote stick, have the sensorboard defined as a device and a vario gauge display to see it is working and have a (very small) map loaded for zooming in/out, getting a list of alternates etc.

All other settings could (should...) probably be removed.

linuxianer99, 2026-03-18:
Is this issue still relevant ?