Openvario/meta-openvario #343: Firmware Upgrade via USB Stick / Online from OpenVario

lordfolken, 2023-03-19:
Hello, 

Firmware upgrades currently mean that we have to remove the SD card from the machine. 
It would be great if the file could be copied to a USB stick and loaded from the stick as a whole file. 
Also an online update feature to the current firmware for the current variant would be fantastic. 

- Folken

mihu-ov, 2023-03-19:
No need to remove the SD card when using ov-revovery.itb (see https://www.openvario.org/doku.php?id=projects:series_00:build:advanced_topics), or is this broken?

mihu-ov, 2024-03-08:
I have tested firmware upgrades using ov-revovery.itb (see https://www.openvario.org/doku.php?id=projects:series_00:build:advanced_topics) and it works with the old images (17119), with the current image (24013) and with the latest image on github (22304). By todays standards the looks are a bit old-fashioned but it still works, loads a new image from USB drive and avoids having to remove the µSD card from the OV.

I am looking forward to improvements and online updates, just wanted to confirm that the recovery implemented by @linuxianer99 more than 8 years ago still does the job :-)

lordfolken, 2024-03-08:
The Opensoar OV image features a much improved routine and upgrade procedure https://opensoar.de/releases/v7.41.21/OV-3.2.21.2/