Openvario/meta-openvario #344: File Manager isn't accessible from Openvario by Stick

lordfolken, 2023-03-19:
If one doesn't have a touch screen installed, its impossible to access the file manager. 

The file manager is convenient as it has an "Update all" button for upgrading all the files currently configured on the device.

lordfolken, 2023-03-26:
https://github.com/XCSoar/XCSoar/pull/1163

lordfolken, 2023-03-26:
i raised the button limit. Please think of other buttons we need to add to the quick menu.

DanD222, 2023-03-26:
> Please think of other buttons we need to add to the quick menu.

Map Display / Airspace / Label visibility (All / None) please.

closed 2023-04-30 by lordfolken

DanD222, 2023-05-16:
@lordfolken I see you've implemented https://github.com/XCSoar/XCSoar/pull/1163/commits/66345d5077fc0c5bf86cf5e5230ac8fa13e1b1b1 - that's useful, thanks.

I'm not sure if you meant to address my post above by it, but what I was after was toggling the `AirspaceLabelSelection` rather than `AirspaceToggleActionName`.

It would also be great if in addition to the QuickMenu, a button to toggle the airspace labels could be added to the Map buttons under the 'Display' button  for easier access on mobile devices.

DanD222, 2023-06-09:
> OK i See what you mean. Indeed that is missing functonality.

Made a separate feature request so that it doesn't get buried here.. https://github.com/XCSoar/XCSoar/issues/1243