Openvario/meta-openvario #348: Fixing CPU speed not really needed

bomilkar, 2023-04-12:
Following up on this PR:
https://github.com/Openvario/meta-openvario/pull/331
I discovered a configuration which is a bit better and still keeps the system stable.
Instead of **fixing** the CPU speed to 720 MHz I **limit** it to 864 MHz. Together with this I **fix** the CPU voltage to 1.3V for all frequencies 864 MHz and below.
I tested this in flight on 2 OVs with 20+ hours each. No issue!

My understanding of the issue:
When the command to change frequency and voltage is issued it takes different amounts of times to settle on the new frequency respectively voltage. Usually it takes many clock cycles until the PLL has settled on a new frequency whereas the voltage "immediately" follows the command. So when the system needs less performance it will throttle down the CPU speed and with it the supply voltage. The time lag between the PLL and the DAC will result in the CPU running at too high clock speed with the already lowered supply voltage.

If the above is true we safely can change the CPU clock as long as we make sure the supply voltage is always sufficient for all possible frequencies. And that is what I did by **fixing** the CPU voltage to 1.3V for all select-able frequencies. (I have a patch for this.)
The advantage: the power consumption is lower when the system needs less performance and the performace is (slightly) higher when it's needed. For instance when the map is very busy or a new map file loads.

And BTW: the old images were running on a kernel which didn't change CPU clock and voltage. The above explains why this used to work.

DanD222, 2023-09-17:
@bomilkar Could you make a PR with your patch please? I’ve been observing some lagginess to joystick inputs in one of the OV forks recently, so maybe this could help a bit.

bomilkar, 2023-09-18:
@DanD222 The patch file is here:
https://xcs.fliegerclub-moosburg.de/share/patches/OV/0001-relax-CPU-clock-and-voltage-constraints-to-provide-h.patch
Although I doubt this will make the joystick respond noticeably faster.

DanD222, 2023-09-18:
OK, thanks Ronald.. good to have a reference to your patch in any case.

bomilkar, 2023-09-18:
You are welcome!
Let me know if it works for your problem, I can then post a PR. Send me email if you think I can help.

bomilkar, 2023-09-18:
As requested: PR #362

closed 2023-09-18 by bomilkar