Openvario/meta-openvario #351: No package 'fmt' found

bomilkar, 2023-04-28:
Aparently XCSoar needs libfmt-dev since a few days now.
See Developer Manual https://xcsoar.readthedocs.io/en/latest/build.html
This package is not in the docker image.

Is there a way to add the package to the existing image? (The equivalent of "apt install libfmt-dev" ?)

bomilkar, 2023-04-29:
Still misses package 'fmt'. This is the error message in /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/log.do_compile.1665518

DEBUG: Executing shell function do_compile
NOTE: make -j 4 HOSTCC=ccache gcc  HOSTCXX=ccache g++  AR=arm-ovlinux-linux-gnueabi-gcc-ar RANLIB=arm-ovlinux-linux-gnueabi-gcc-ranlib CXX=ccache arm-ovlinux-linux-gnueabi-g++  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/recipe-sysroot CC=ccache arm-ovlinux-linux-gnueabi-gcc  -mthumb -mfpu=neon -mfloat-abi=hard -mcpu=cortex-a7 -fstack-protector-strong  -O2 -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security --sysroot=/workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/recipe-sysroot AS=arm-ovlinux-linux-gnueabi-as  LD=arm-ovlinux-linux-gnueabi-ld --sysroot=/workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/recipe-sysroot  STRIP=arm-ovlinux-linux-gnueabi-strip CCACHE= DEBUG=n DEBUG_GLIBCXX=n ENABLE_MESA_KMS=y GLES2=y GEOTIFF=n
**Package fmt was not found in the pkg-config search path.
Perhaps you should add the directory containing `fmt.pc'
to the PKG_CONFIG_PATH environment variable
No package 'fmt' found**
build/test.mk:2322: *** library not found: fmt.  Stop.
ERROR: oe_runmake failed
WARNING: /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/run.do_compile.1665518:177 exit 1 from 'exit 1'
WARNING: Backtrace (BB generated script):
        #1: bbfatal_log, /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/run.do_compile.1665518, line 177
        #2: die, /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/run.do_compile.1665518, line 161
        #3: oe_runmake, /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/run.do_compile.1665518, line 156
        #4: do_compile, /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/run.do_compile.1665518, line 151
        #5: main, /workdir/tmp/work/cortexa7t2hf-neon-ovlinux-linux-gnueabi/xcsoar-testing/git-r13/temp/run.do_compile.1665518, line 190

lordfolken, 2023-04-30:
Ah, this is openvario's image not "mine".

lordfolken, 2023-04-30:
https://github.com/Openvario/ovbuild-container/pull/6 potenially.

MaxKellermann, 2023-05-01:
This can be solved by adding `fmt` to the list of build-dependencies for 7.29. Can't get any simpler, can it?

bomilkar, 2023-05-01:
I confirm https://github.com/MaxKellermann/meta-openvario/commit/5a49b2e3425a044f7b67980d9cb95e22de6b281f works. Indeed very simple.

closed 2023-05-01 by bomilkar