Openvario/meta-openvario #353: WiFi/BT

hpmax, 2023-05-09:
Is there a comprehensive list of WiFi and BT USB adapters or chipsets that are supported (or preferably combo WiFi/BT adapters) or at least

bomilkar, 2023-05-14:
Such a list would be very useful.
However, most dongles don't reveal brand or type unless you plug it in and read the device's VID and PID.
For instance: I have a "EDIMAX 150 MBit/s 802.11bgn Nano" working fine. On the outside it just says EDIMAX. The IDs are VID_7392 & PID_7811.
Maybe we should build a list of devices which are known to work (with a given OV image).

hpmax, 2023-05-14:
From what I can tell, any RT5370 based WiFi adapter works by default.  The question is how do we or can we add more drivers for other chipsets (particularly WiFi/BT combo adapters).