Openvario/meta-openvario #361: Screen blanks with Image 22304.

RN-Cook, 2023-09-08:
I ran OV image 17119 for several years (on a Stefly kit with sensorboard). I had a problem which seemed to relate to cockpit temperature. On cool days all was OK. On hot days the audio produced 2 or 3 loud clicks per second and the vario tones were strange. As I got higher into cooler air the problem often cured itself. I put up with this.

I have now updated to Image 22304 and have a new more serious problem (plus the old one)  The screen blanked after about 3 hours flight and could only be restored by turning it off and on again. After about 30 mins it did it again. 

I repeated this in a ground test. The air temperature was 27/28 C. The speaker started clicking immediately. I left the glider for just over 3 hrs  When I returned, the screen had blanked out (dark grey with a slightly lighter stripe at the top). The clicking was still there. Blowing in the TE tube showed the audio vario was not working, and the OV did not respond to the stick-top controller. As in the air I had to switch the OV off then on again. After about 30 mins it blanked out again.

Anyone got any suggestions please? Is there a fix? Should I go back to some earlier build?

bomilkar, 2023-09-20:
@RN-Cook if you are keen on trying the latest pre-release I can build one for you if you tell me which flavor (display) you need. I'd be interested to hear how it behaves. There are some significant fixes since 22304 which should apply to your issues.

RN-Cook, 2023-09-20:
Hi Ronald

Thank you. I would be happy to try it. I built my OpenVario from a Stefly
kit with a 7" screen (not a touch screeen) with Stefly sensor board and
stick-top remote. Is this all the information you need?

Regards Rob Cook

bomilkar, 2023-09-21:
@RN-Cook send me an email and I'll send you the download link.
I assume you have a CH070 display.
ronald.niederhagen@gmail.com

RN-Cook, 2024-07-19:
I have (at last) installed the image that Ronald gave me. Thank you Ronald.  I flew with it yesterday on a 28° day (hot by UK standards). The OV crashed after about 2.5 hours. I switched it off and on again. It came back with a fixed constant lift value in the 1000’s, and no audio output. Later it crashed again. I switched it off and on again. This time it came back with no controlled airspace showing, just TP's and terrain. I decided to fly home.

There are OV's in our club's single seaters, and several private gliders have them too. No-one else has reported this problem.

I didn't have this problem with OV image 17119, but I still wonder if this is simply a temperature issue. Should I add heat-sinks to key internal components? Can I make ventilation holes in the casing (without effecting RX interference)? Should I add a cooling fan?

Regards Rob Cook

bomilkar, 2024-07-22:
Rob,

this looks like a temperature issue. I flew at higher temperatures (~35°) w/o issues, but I have no sensor board, just a plain 7" SteFly without any heat sinks or holes. Just the aluminum box. However, I had one issue several years ago when the glider was parked on the ground for a couple of hours, during noon time in full sun exposure w/o a canopy cover: it froze 5 minutes after takeoff. Switched it off and on again and it came back fully operational. I avoided these super heated conditions (on the ground) and the issue never appeared again.
I would not drill holes in the casing or add a cooling fan. There is the risk of picking up dust which makes things even worse.  A good heat sink may help, but what are the "key internal components" ??
You can try a newer image but I doubt this would make a difference (let me know). If it remains a singular event, I wouldn't bother too much. Is it showing the airspace correctly? That might indicate a corrupt file system and more than just the airspace file is affected, in which case I would write a backup to the SD card. If you have taken a copy, otherwise start from scratch.

RN-Cook, 2024-07-22:
Thanks Ronald

I think the critical component is the cubie board processor. There are heat sinks online aimed at cubie boards.
http://cubieboard.org/2013/09/01/2-in-1-heatsink-package-for-cubies/
https://elediy.com/products/ultrathin-heat-sink-for-cubieboard
https://www.joy-it.net/en/products/RB-heatsink2
I also found a suitable fan, but not sure it will fit inside the housing. https://www.joy-it.net/en/products/RB-heatsink2

But perhaps the sensor board is the critical factor.  I am not sure if other users at my club have the sensor board. Perhaps fit a heat sink to its main chip?
But all of this is a job for the end of season when I can take the OV out of the panel. For the moment I have reinstalled image 17119 which did not crash on me when I was using it before - but perhaps the UK did not have any hot days back then.