Openvario/meta-openvario #366: Cubieboard 6/7

hpmax, 2023-10-15:
I just got my hands on a Cubieboard 7, under the impression it was a drop-in/backward compatible replacement for the Cubieboard2 except with a much more advanced processor, more memory, and WiFi/BT built in.  I powered up and saw nothing on the screen.  Does anyone have any idea what's wrong?  Bad CB7, hardware incompatibility, software incompatibility?

lordfolken, 2023-10-16:
"best to make it be compatible with CubieBoard2 in hardware, such as board shape, board size, interface locations, and the function of expansion pins. Even so a small number of the expansion pins are not identical in function, since these two boards adopt the different main chips."

http://www.cubietech.com/product-detail/cubieboard6/

Plus the whole driver and kernel image has been focused on the cubieboard2. So i would be very surprised if it booted. 

Personally i would focus on rasperry pis / hdmi connection. There is no advantage on using an obscure platform.

hpmax, 2023-10-18:
First, I bought the Cubieboard 7 because I believed it was fully backward compatible.  It appears I may have been wrong.

Second, I agree that Cubieboard is an "obscure platform", however I'm invested in the sense that I have a Stefly OpenVario.  Everything about it is based around the Cubieboard formfactor.  I'd need a different adapterboard, I'd need different holes in the case, I'd need a new display or an HDMI->LVDS converter.  The lack of an LVDS output on the Raspberry Pi is kind of mind boggling to me considering that most barebones displays are LVDS not HDMI.

Nonetheless, the question is... is there an actual hardware or software incompatibility between the two boards?