Openvario/meta-openvario #367: Upgrade XCSoar to v7.40 - SRCREV

DanD222, 2023-11-04:
XCSoar 7.40 has been released https://github.com/XCSoar/XCSoar/releases/tag/v7.40, so it might be worth updating the XCSoar version used by OpenVario - does anyone know what the SRCREV value in `meta-ov/recipes-apps/xcsoar/xcsoar_7.xx.bb` should be?

I haven't had luck with figuring this out from the past updates unfortunately.

linuxianer99, 2023-11-08:
The easy, but maybe not correct way would be to leave it as it was. Try to build the recipe ... bitbake will bring an error that checksum is not correct and shows also the expected ;-) Copy it ... done ...

DanD222, 2023-11-08:
Maybe I don't understand, but wouldn't that keep building the version that the SRCREV points to? How will it know that I'm trying to build 7.49 and not 7.35?

Blaubart, 2023-11-13:
I asked the same question at the forum:
https://forum.xcsoar.org/viewtopic.php?f=30&t=4142#p23647

DanD222, 2023-11-13:
Thanks! For reference:

> the SRCREV is the GPG-signed release tag. Try:

`~/git/xcsoar$ git rev-parse v7.40`
c42bc03d6c4e06268ed891bc2a7572537ea81de6

or
`~/git/xcsoar$ git show c42bc03d6c4e06268ed891bc2a7572537ea81de6`

closed 2023-11-13 by DanD222