Openvario/meta-openvario #371: XCSoar crash with Libinput warning message

mihu-ov, 2024-03-08:
I installed a self compiled image (24013, still current, with @kedder OV Shell) on 2 OVs in our club. On both OVs XCSoar crashes with libinput warning messages "lagging behind by XXms, your system is too slow". 

On the first OV, this happens sometimes. On the second OV it happens frequently, and also vario sound stops after a few seconds (both variod and XCSoar generated). The vario itself (sensor data and vario gauge) keep working.

I took the µSD card from the second club OV, installed it in my OV sitting on my desk  and tried to replicate these issues. but everything works fine here. I noticed CPU load shown in the XCSoar infobox is lower on my OV. All 3 OVs are the same build and same age, the only differences are no touchscreen installed on mine (removed) and no serial devices connected on my desk.

Could the (ageing) touchscreen generate "ghost" inputs overlading the processor? Any hints what to check or filter? I would prefer not having to physically disconnect the touch for testing due to a broken arm.

mihu-ov, 2024-03-09:
[image]

This is the screen after XCSoar crashes

lordfolken, 2026-03-22:
Well i had a look at libinput.cpp and there the events are not being destroyed after consumption by the handler. This is required according to the API docs. So this is a valid issue. 
However not sure about it being the cause of the crash.