Openvario/meta-openvario #375: OpenVario DS2.The reachable terrain line is not displayed.

arpass, 2024-05-22:
Hello friends,

I have encountered an unexpected problem with my OpenVario DS2. The reachable terrain line is not displayed. Additionally, the runways do not change color to green, although they do turn yellow. However, the required glide altitude is displayed correctly. The polar diagram for the Std.Libelle glider is activated. The flight altitude and speed display also work correctly. Could you please help me identify where to look for the issue?

Thank you in advance!

Scumi, 2024-05-22:
I can confirm those issues. It appears to be an issue with the UNIX target of XCSoar as the same issues appear on my linux laptop. We should open two bug reports on the XCSoar/XCSoar main repo.

Scumi, 2024-05-23:
So apparently an old compiler is responsible for the bug. Can we update the compiler used for the OpenVario build to its latest version (>= gcc-12/g++12)?