Openvario/meta-openvario #377: SSL Certificates invalid

lordfolken, 2024-07-25:
i installed current testing image today on my openvario.
While everything works as expected, i found too problems after createing a wifi connection with connman:
 - when downloading the repository, the file contained the http://download.xcsoar.org/repository 302 message, which redirects to https://download.xcsoar.org/repository (i modified download.xcsoar.org, to no longer enforace https)
 - when trying to download a rasp source, it complained that it couldn't find the issuer ca for thermalmap.info

wget <urls> on the console seems to work fine. 
Downloads work fine on amd64 plain xcsoar.

DanD222, 2024-08-21:
A workaround for this on existing OpenVario devices is to:

1) Connect the OpenVario to the Internet using e.g. an Android phone with USB Tethering enabled, via a USB cable
2) Exit to the shell
3) Run the `update-ca-certificates -f` command
4) Reboot the OpenVario

WeGlide integration for example will then work over WiFi again

lordfolken, 2024-08-22:
update-ca-certificates shouldn't need an internet connection? 

All this does is build all the keystores (java,mono, systemwide ca-certificates,etc) from /etc/ssl/certs/ /usr/share/ca-certificates. 

This seems to be executed upon installation of the package:

https://github.com/openembedded/openembedded-core/blob/hardknott/meta/recipes-support/ca-certificates/ca-certificates_20211016.bb#L69

There is also a ca-certificates file in the resulting image:
-rw-r--r-- 1 root root 191K Mar  9  2018 ca-certificates.crt

But if its from 2018...

lordfolken, 2024-08-24:
So I tried it today. The CAs are too old. For example thermalmap.info wouldnt connect.

I downloaded the Mozilla cert store from https://curl.se/docs/caextract.html

Moved that into place of /etc/ssl/certs/ca-certificates.crt.

After restarting timesyncd to sync the time, I could download rasps from thermalmap.