Openvario/meta-openvario #391: New Openvario Image / Yocto Scarthgap?

mihu-ov, 2025-03-27:
Are there any plans to release current Openvario images for the start of gliding season in the northern hemisphere?
I was able to compile current master but failed with master-next.

Scumi, 2025-03-27:
My master-next compiles just fine. I think you have to pull the latest docker image.

I fully support a new release and would create a PR upping the XCSoar version on master, if master-next is not ready yet.

mihu-ov, 2025-03-27:
I would appreciate it, and probably many other OV users as well!

linuxianer99, 2025-04-04:
Hi All :-)
Think that should be possible to build an image .. but ...

There are some breaking changes:

- Versioning is not longer <year, dateofbuild> but just a number based on Yocto versioning. So i have no clue, how ovmenu-ng, the update stick, etc are handling this.

Proposal: 
I can build an "release" image, and somebody did some tests on this ?
I would not say it is an official release .. mabye we break some Openvarios doing so ...

What so you think ?

mihu-ov, 2025-04-04:
Hi Timo,

This would be great! I could test this weekend but am away all next week and also the week after.
Is there a way to get the driver for Becker radios included?

@MaxKellermann 
Do you still use the Openvario I sent you? If yes, that´s great! If you don´t use it anymore, would you be able to send it back to me? My OV is installed in my glider and once the season starts it´s getting hard to test new images.

linuxianer99, 2025-04-09:
@mihu-ov : What is this driver about ?
Is not just a serial port ??

mihu-ov, 2025-04-09:
On the radio side, it is a RS422 serial port. In XCSoar there was a pull request for implementation of a driver for Becker radios that was rejected, as far as I remember mostly because the documentation is not made public by Becker. The driver is included in @August2111 OpenSoar releases so he can probably fill in the details.

mihu-ov, 2026-03-29:
Any chance for a PRE-1 image @linuxianer99 ? Will it have XCSoar, OpenSoar or both?

Scumi, 2026-04-14:
Hey Uwe :-)

My goal was not to re-invent the wheel, but improve upon it. The OpenSoar-derived commits were not fully working when trying to use XCSoar, which needed to be fixed for XCSoar users. After that I improved the existing scripts while trying to not regress functionality (we're talking #431 here I suppose).

Lets have a look at it:

1. New partition layout overriding p3  
This is a very valid point, and something I hadn’t fully considered. My intention was to simplify the workflow by allowing all files to be placed on the SD card right after the build, which also helps with recovery scenarios. However (as you pointed out) this would overwrite p3 and break existing setups. I’m happy to revert this change - especially since the long-term direction is flashing via USB/network anyway, so this was more of an interim solution anyway.

2. XCSoar being installed in parallel to OpenSoar  
This didn’t seem to be the case in the scarthgap branch. Before #431, there is an exclusion coded based on a DISTRO_FEATURES-setting. Also, when XCSoar was installed and enabled, it tried to work with the OpenSoar menu, which does not work due to the different approach to OpenVario control. Also, requiring users to manually change the boot entry goes against the goal of making this accessible to non-technical users. For now, I’d prefer to keep the app selector and separation logic until XCSoar no longer depends on its own dialog/menu.

On the broader point: I think the disconnect largely comes from parallel work in forks that wasn’t fully visible from the main repo (at least not to me). I’ve been basing my work on what’s available here, trying to progress incrementally while aiming to avoid regressions.

I do agree that better alignment would have reduced duplicated effort. Happy to align going forward.

mihu-ov, 2026-05-09:
@Scumi In https://github.com/Openvario/meta-openvario/discussions/439#discussioncomment-16843872 you wrote you are flying with 5.0.9-1-PRE1.

Do you fly with the standard 5.0.9-1-PRE1 image, or with any changes included?
For me, both standard and testing image boot into OpenSoar, there is no selection of OpenSoar or XCSoar. If I manually edit /boot/conifg.uEnv to main_app=xcsoar it boots to xcsoar, and when I quit XCSoar it exits to shell. There is no way to set up WiFi or anything. 

After the new 5.0.9-1-PRE1 release (should this be -PRE2?) I changed to OpenVario directory, did "git pull" and continued according to instructions in https://github.com/Openvario/OpenVario/blob/main/README.md
Or do I need to delete everything and compile again?

linuxianer99, 2026-05-18:
@August2111 @mihu-ov 
Thanks for all the feedback. Let me create some issues and fix the things ...

@Scumi 
I am also a fan of working together and not wasting time :-) How can we align  ?? And which topics are you seeing most important ?